lancaster-university/microbit-dal #115: AnalogOut (PWM) should be available on all digital enabled pins

finneyj, 2016-04-12:
is currently restricted to those with an AtoD converter - not needed for PWM operation.

sh1boot, 2016-11-27:
What actually needs to be done for this?  I'm currently using PWM output on pins 15 and 16 by simply bypassing MicroBitIO.cpp and instantiating the pins with `PIN_CAPABILITY_AD`; which is a lie, but it gets me past the corresponding test in `MicroBitPin::setAnalogValue()`.

Unless there are pins which cannot support PWM output, isn't it simply a matter of deleting that test?

jaustin, 2016-11-28:
@finneyj I'm guessing that fixing this needs a bit of work to decouple PWM out from the pins that can do ADC when configured as input, is that right? 

If so, should we sketch out a design and mark as up for grabs?

jamesadevine, 2016-11-29:
@jaustin Yes, there needs to be a separation between Analog generally, and specific Analog operations.

The enum currently looks like this:

```cpp
enum PinCapability{
    PIN_CAPABILITY_DIGITAL = 0x01,
    PIN_CAPABILITY_ANALOG = 0x02,
    PIN_CAPABILITY_AD = PIN_CAPABILITY_DIGITAL | PIN_CAPABILITY_ANALOG,
    PIN_CAPABILITY_ALL = PIN_CAPABILITY_DIGITAL | PIN_CAPABILITY_ANALOG
};
```
Although unnecessary, the perfect patch would be to bring out IN/OUT for each mode:

```cpp
enum PinCapability{
    PIN_CAPABILITY_DIGITAL_IN = 0x01,
    PIN_CAPABILITY_DIGITAL_OUT = 0x02,
    PIN_CAPABILITY_DIGITAL = PIN_CAPABILITY_DIGITAL_IN | PIN_CAPABILITY_DIGITAL_OUT,
    PIN_CAPABILITY_ANALOG_IN = 0x04,
    PIN_CAPABILITY_ANALOG_OUT = 0x08,
    PIN_CAPABILITY_ANALOG = PIN_CAPABILITY_ANALOG_IN | PIN_CAPABILITY_ANALOG_OUT,
    PIN_CAPABILITY_ALL = PIN_CAPABILITY_DIGITAL | PIN_CAPABILITY_ANALOG
};
```
I think this allows for maximised capability, you could thereby specify pins as Digital or Analog only, or any combination of the above.

The logic would remain simple in `MicroBitPin` as now you would only have to check against the flag that specifically relates the desired operation:

```cpp
int MicroBitPin::setServoPulseUs(int pulseWidth)
{
    //check if this pin has an analogue mode...
    if(!(PIN_CAPABILITY_ANALOG_OUT & capability))
        return MICROBIT_NOT_SUPPORTED;

    .....
```

whaleygeek, 2016-11-29:
Related PXT ticket: https://github.com/Microsoft/pxt/issues/816

whaleygeek, 2017-02-11:
Note, this seems to still not be working in PXT, Peli has directed us over to here:

https://github.com/Microsoft/pxt/issues/611

whaleygeek, 2017-02-11:
[image]

Which one of these numbers is the DAL version?

pelikhan, 2017-02-11:
For that, you need to look at the pxtarget.json file in PXT-microbit repo for that release.

[screen shot 2017-02-11 at 17 37 34]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcloud.githubusercontent.com%2Fassets%2F4086785%2F22855984%2Ffdc5833c-f080-11e6-966f-f007b44be056.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C70cb44d0c6814856fd9108d452a4ea8c%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636224315639371896&sdata=Um5xQAVP6INVivDGmi6P0szv%2FQFz9IumkEq46%2BPvM1w%3D&reserved=0>

whaleygeek, 2017-02-11:
@pelikhan Which bit of this file please?

[image]

whaleygeek, 2017-02-11:
@pelikhan Oh, hang on? You guys have two repos? We only log issues against the core PXT but the code is actually built from pxt-microbit?

https://github.com/Microsoft/pxt-microbit/blob/master/pxtarget.json

whaleygeek, 2017-02-11:
So, this is the link I presume? Thanks.

[image]

pelikhan, 2017-02-11:
Yup v2.0.0-rc7 I believe this is the latest release.

[screen shot 2017-02-11 at 17 48 06]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcloud.githubusercontent.com%2Fassets%2F4086785%2F22856065%2F4b77c4d6-f082-11e6-9032-e91ad6ec8024.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C6e76be4eed08454d030608d452a63397%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636224321176257729&sdata=LEdaQ0v0hg4ZN0oZmkMVNWnrkHszfDo4BR2wIJEJJVc%3D&reserved=0>

luniki, 2017-02-17:
@sh1boot Could you provide an example how you are using PWM output on pins 15 and 16? I would like to try this on my own too.

jamesadevine, 2017-02-17:
@luniki You could test my PR: #264

DavidWhaleMEF, 2017-02-27:
Is there any progress on this at all do you know? It's holding up a bug fix in PXT that we've got a long queue of people waiting for - ta!

https://github.com/Microsoft/pxt/issues/611

jaustin, 2017-02-27:
> Microsoft/pxt#611 <https://github.com/Microsoft/pxt/issues/611>

And it's a change to make in DAL before RC...

J

finneyj, 2017-02-28:
will do today

finneyj, 2017-03-01:
Quick update.  All changes made. Just want to run a final test suite when i'm a bit less tired in the morning... I'll tag a new release early Wednesday AM UK time.

Jaqster, 2017-03-01:
Hi guys.  We're releasing RC today.  We can work in the DAL changes after RC (it's not a breaking/UI change).

jamesadevine, 2017-03-01:
@Jaqster The fix for this bug is in `v2.0.0-rc8` all tagged and ready to go :smile:

closed 2017-03-01 by jamesadevine