lancaster-university/microbit-dal #118: Need BLE connection state change events

bluetooth-mdw, 2016-04-14:
Any BLE service (e,g, the new UART service) where use can be driven by micro:bit app code will need to know when a client connection has been established or lost. So some microbit events to listen to would be useful.

Suggest event called BLE_CONNECTION_STATE and values called BLE_CONNECTION_ESTABLISHED and BLE_CONNECTION_LOST or something along those lines.

So in the context of the UART service for example, I'd listen for connection established events like this:

`uBit.messageBus.listen(BLE_CONNECTION_STATE, BLE_CONNECTION_ESTABLISHED, onBleConnectionEstablished); 

void onBleConnectionEstablished(MicroBitEvent e) {
   // do something involving Bluetooth
}`

jamesadevine, 2016-04-14:
@bluetooth-mdw something @finneyj and I have spoken about, but have never implemented. It is insanely simple, and extremely powerful :smile:

Thanks!

finneyj, 2016-04-14:
yep - agreed. There's little cost to raising events (as the MessageBus receiver driven), so the more the merrier really.

@bluetooth-mdw We'll get to this, but if you'd like to submit a pull request that would be awesome. All you need to do is create a MicroBitImage whenever you want to raise an event, where id and value are whatever you want to raise:

```
MicroBitEvent(id,value);
```

bluetooth-mdw, 2016-04-15:
You trust me to do this?! ;-)

OK I'd be delighted. On my list.....

finneyj, 2016-04-15:
:-) Awesome - thanks @bluetooth-mdw.

If you're licensed to build a Bluetooth Buggy, you're licensed to build a Buggy Bluetooth. :-D

bluetooth-mdw, 2016-04-15:
Puntastic!

bluetooth-mdw, 2016-04-15:
Pull request sent: 

https://github.com/lancaster-university/microbit-dal/pull/120

Be amazed at:

https://drive.google.com/file/d/0B9Z54U0dfwbFN0Y0RVVROS1DQWM/view?usp=sharing

closed 2016-04-15 by bluetooth-mdw