lancaster-university/microbit-dal #121: Can't download 2.5.0+mb2

carlosperate, 2016-04-15:
using the offline toolchain, at some point today I started getting the following error on `yt up`:

```
info: get versions for ble
info: download ble@v2.5.0+mb2 from GitHub lancaster-university/BLE_API
critical: GET https://codeload.github.com/lancaster-university/BLE_API/legacy.tar.gz/v2.5.0%2Bmb2 failed with status 404
```
Trying to access https://codeload.github.com/lancaster-university/BLE_API/legacy.tar.gz/v2.5.0%2Bmb2 from a browser results in a "not found" page being displayed.

I wasn't quire sure if this is the right place to report this, as the BLE_API repository fork has the GitHub issues deactivated.

carlosperate, 2016-04-16:
Woke up this morning and the link is working again. I'm not quite familiar with how codeload.github.com works, is that where files are stored when a github release is made?

jamesadevine, 2016-04-16:
@carlosperate I would probably conclude that this was perhaps a github outage? Looking at the stats though - there were no reported outages.

It may be better to open an issue on the yotta repo:

https://github.com/ARMmbed/yotta

They would definitely no more about this there!

carlosperate, 2016-04-16:
Having a look around it, yotta takes the github tag from https://github.com/lancaster-university/microbit-dal/blob/c064eee3e7e34c2f0b745f37d87f96ea88779b98/module.json#L18 , so as you say, it looks like a github outage, even if it wasn't really reported. I don't think there any need to look into it in more depth.

Thanks!

closed 2016-04-16 by carlosperate

jamesadevine, 2016-04-16:
Thanks for letting us know @carlosperate! :smile: