lancaster-university/microbit-dal #122: Custon configuration support

pelikhan, 2016-04-16:
The instructions in sample state that one should change some header files from the DAL to get radio working. 

https://github.com/lancaster-university/microbit-samples/blob/master/source/MicroBitSamples.h#L53

Chaning files under the `yotta_modules/...` is probably not a great idea. Is there a better way to do this - the ``MicroBitCustomConfig.h`` file way?

matthewelse, 2016-04-16:
@pelikhan @jamesadevine you might find this: http://yottadocs.mbed.com/reference/config.html useful, which explains how to use Yotta's built in config system.

jamesadevine, 2016-04-16:
@pelikhan thanks for raising this :smile:

@matthewelse I think we need to be open in terms of the way configurations are defined and certainly the semantic meaning behind those configuration options. Although yotta has this config feature built in, it does muddy the water in terms of the #defines it generates.

We also have to consider the potential of other offline build systems targeting microbit in the future, and it would probably be objectionable from their point of view to have yotta specific #defines.

Do you know if yotta allows custom prefixes for #defines?

finneyj, 2016-04-16:
Thanks @matthewelse @pelikhan 

@matthewelse - there are quite a few config options for microbit-dal so in an earlier version we used the YOTTA_CONFIG system to specify a local pre-include file where the developer could specify option to override the defaults set in MicroBitConfig.h... This options was simply overlooked when we refactored for the 2.0.0 version, so I was thinking we'd resurrect that (as @jamesadevine says - this is a bit more independent of build system - almost anything can handle a pre-include) ?

Open to alternate suggestions if anyone has better ideas though!

mmoskal, 2016-04-16:
I would vote for keeping the customconfig.h but this is just because I'm lazy ;)

finneyj, 2016-04-16:
that's a fine reason, not to be underestimated!

pelikhan, 2016-04-16:
customconfig.h worked pretty well so far, sounds like a great star.

remay, 2016-04-17:
I'm not a yotta expert ... but I see a couple of benefits to using the yotta configuration system (at least for some of the configuration values):

1. Parameters can be set in the targets.  For example this would allow the standard (S110) target to specify different memory limits to the S130 target, allowing both to work without having to customise anything in a project.
2. The yotta defines are globally available to all modules being built.  The CustomConfig.h values were only available to the microbit-dal part of a build.  (I ran into this in the old codebase when including microbit.h in my main.cpp and printing values for testing purposes.  It didn't print the expected (customised values.  I considered this surprising 'action at a distance')  I suspect many people will fall for this.

pelikhan, 2016-04-18:
Do you have a pointer to the docs about these global defines? IMO, this is quote important -- specially to disable BLE -- and the yotta approach is not obvious. An example in the samples would be great!

remay, 2016-04-18:
Yotta config data is explained here: http://yottadocs.mbed.com/reference/config.html

pelikhan, 2016-04-18:
Not so easy: http://yottadocs.mbed.com/reference/config.html#use-in-code

It seems that yotta prepends all generate defines with ``YOTTA_`` so we would not be able to override ``MICROBIT_`` defines. This would require a massive renaming in the DAL.

jaustin, 2016-04-18:
I don't think it would require a renaming - we could do it with an integration header. I often use yotta config along with a spearate header to clean things up

#define MY_CONF YOTTA_CFG_XXX

We could even only include this header when building with yotta, leaving other build systems to do their own equivalent. It does make using config.json files and sharing them nice, and as @remay points our, the ability to have things defined in the targets is pretty sweet - it means I could make a custom target for my app and share *that* with you instead of telling you to set a bunch of config.

pelikhan, 2016-04-18:
> the ability to have things defined in the targets is pretty sweet 

Humm, are we trying to define thousands of targets? it seems that with this technique, you'd get a new one for every app with slightly different settings. Currently, a yotta user has to write "yotta target ..." once and forget about it. Constantly changing target seems like an overkil.

* Can't we just let the user specify a file that should preincluded instead of the JSON->C++ name mangling that does not work for everyone?
* Would it be possible for targets to override the ``YOTTA_CFG_`` prefix in ``module.json`` with something else?

pelikhan, 2016-04-18:
I guess my ignorance of the yotta build system is at fault here, does anyone has a working example where DAL properties are being globally defined **without** defining a new target? This is what the previous solution gave us. Defining new targets seems like a regression.

jaustin, 2016-04-18:
So, yotta config can be set by targets, or by your local config.json
So, if sharing an app with you, I can make a config.json in that application that sets all the required config for me.

The targets thing is more for having something like a "no bonding" target or something that easily lets people set specific, globally useful config. We *sort of* have precedent here with the 'nosd' config, etc.

pelikhan, 2016-04-18:
Ok great local config.json works. Next, the DAL would have to add a "mapping" file that redefines every YOTTA_CFG_ constant as MICROBIT_ (at least the ones that matter). Is that a correct assumption?

autopulated, 2016-04-18:
Hey, chiming in on request of @jaustin:

There are a few different things you could do here to take advantage of the yotta config system:

 1. A mapping header between YOTTA_CFG_XXX and MICROBIT_XXX definitions
 2. Add a [pre-build script](http://yottadocs.mbed.com/reference/module.html#scripts) which reads the config data (which is available to scripts as a merged JSON file), and generates a header with the right #define MICROBIT_XXX values in.
 3. You could encapsulate such logic to make things from the yotta config available under a different namespace as a script in a separate yotta module that other people could use for the same purpose. For example you could add `"$exposeDefinition": true` or similar markers to the config, and have a module traverse the config and pull out all the values from positions in the config with that marker, and #define them as un-namespaced definitions.

In all cases the new header should be explicitly included somewhere (it might replace a header which is normally empty, or which contains default config). (The fact that the yotta config values are pre-included in everything being built was not the best choice in retrospect, so I wouldn't recommend adding another pre-included header, even though you might be able to use CMake to do that).

The config system was designed to make things like 3. possible, with the idea that people might want to even embed sections of the config information to be available at runtime, with overridable tags in the config marking which sections should be processed. You could, for example, have one module which embeds tagged parts of the config as runtime-accessible CBOR, or a different one which embeds it as runtime-accessible JSON.

Regarding better documentation of yotta config values, the proposal is to use schemas (possibly with comments), see this open issue: https://github.com/ARMmbed/yotta/issues/325

jamesadevine, 2016-04-18:
Thanks @autopulated a lot of food for thought :smile:

Simplest path to success looks like a mapping header, however I am intrigued by the idea of a pre-build script...!

@autopulated / @jaustin got an example of a pre build script in action?

autopulated, 2016-04-18:
I just put together a [simple version](https://github.com/autopulated/expose-defs) of the expose-defs idea mentioned :)

pelikhan, 2016-04-18:
Could we have a mapping file added to the DAL? I fear that the 'expose-defs' is yet another dependency that's not quite needed assuming we have a mapping file.

jamesadevine, 2016-04-18:
@autopulated thanks for the example.

@pelikhan it definitely looks like a mapping is the best way to go!

pelikhan, 2016-04-18:
James, could add a mapping for disabling BLE to unblock us?

jamesadevine, 2016-04-19:
@pelikhan 8875dc35d2f5aa3d891c318ad85ec38734c93d79

```
{
    "microbit-dal":{
        "bluetooth":{
            "enabled": 1,
            "pairing_mode": 1,
            "private_addressing": 0,
            "open": 0,
            "whitelist": 1,
            "advertising_timeout": 0,
            "tx_power": 0,
            "dfu_service": 1,
            "event_service": 1,
            "device_info_service": 1
        },
        "reuse_sd": 1,
        "gatt_table_size": "0x300",
        "heap_allocator": 1,
        "nested_heap_proportion": 0.75,
        "system_tick_period": 6,
        "system_components": 10,
        "idle_components": 6,
        "use_accel_lsb": 0,
        "min_display_brightness": 1,
        "max_display_brightness": 255,
        "display_scroll_speed": 120,
        "display_scroll_stride": -1,
        "display_print_speed": 400,
        "panic_on_heap_full": 1,
        "debug": 0,
        "heap_debug": 0,
        "stack_size":2048,
        "sram_base":"0x20000008",
        "sram_end":"0x20004000",
        "sd_limit":"0x20002000",
        "gatt_table_start":"0x20001900"
    }
}
```

You can also do a subset of config options:

```
{
    "microbit-dal":{
        "bluetooth":{
            "open": 1
        },
        "debug":1
    }
}
```

pelikhan, 2016-04-26:
Looks like this issue is fixed.

finneyj, 2016-04-27:
Thanks Peli. closing.

closed 2016-04-27 by finneyj