lancaster-university/microbit-dal #123: storage: document limitation?

pelikhan, 2016-04-16:
Could you add more "limitation" notes about the storage? max size of key, value, computational cost of lookup / update.

It seems that the flash is rewritten on each update so updating the value in a tight loop is probably not the best of ideas if one has many keys?

jamesadevine, 2016-04-16:
@pelikhan http://lancaster-university.github.io/microbit-docs/ubit/storage/#operations Do you feel it would be better to add this to the header files then?

Yes, flash is rewritten on each update. This storage module is intended to be purely used for lightweight, occasional operations, @alexk2k12's file system would be a better solution for large amounts of data.

mmoskal, 2016-04-16:
Also, the doc say you have to free KeyValuePair, but the example doesn't.

jamesadevine, 2016-04-17:
@mmoskal Great point, that is an example of bad practice 😉  although that example doesn't strictly need it. Will update, thanks for pointing that out!

jamesadevine, 2016-04-19:
@mmoskal @pelikhan Documentation has been updated on the docs-site, do you feel anything else needs to be done here?

mmoskal, 2016-04-19:
Looks good!

closed 2016-04-19 by jamesadevine