lancaster-university/microbit-dal #125: UART service readUntil not working as expected when specified buffer size fully used

bluetooth-mdw, 2016-04-18:
With this code:

`   
  uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_CONNECTED, onConnected);
  uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_DISCONNECTED, onDisconnected);
 uart = new MicroBitUARTService(*uBit.ble, 32, 32); 
`

and

`void onConnected(MicroBitEvent e)
{
    
    uBit.display.scroll("C");
    
    // my client will send ASCII strings terminated with the colon character
    ManagedString eom(":");
    
    while(1) {
        ManagedString msg = uart->readUntil(eom);
        uBit.display.scroll(msg);
    }
    
}`

Using nRF Master Control Panel to write to the RX characteristic (i.e. sending data to the micro:bit)

1. Testing with 4 x RX writes from the client app of-> 1234567890 then 1234567890 then 1234567890 then  : 

i.e. final write is a single colon character which is my message terminator for readUntil 

This works. I see nothing on the micro:bit display until the final colon is received and then the whole message, sent in those 4 BLE writes scrolls across the display without the terminating colon. 

However...

2. Test with 4 x RX writes: 1234567890 1234567890 1234567890 1: does not work. 

i.e. here I'm sending a total of 32 octets, the last one being my terminator. I don't see the message 1234567890123456789012345678901 scrolling across the screen. Assume the terminator is not making it into the 32 octet buffer or something like that.

Whole micro:bit app fyi:

`MicroBit uBit;
MicroBitUARTService *uart;

void onConnected(MicroBitEvent e)
{
    
    uBit.display.scroll("C");
    
    // my client will send ASCII strings terminated with the colon character
    ManagedString eom(":");
    
    while(1) {
        ManagedString msg = uart->readUntil(eom);
        uBit.display.scroll(msg);
    }
    
}

void onDisconnected(MicroBitEvent e)
{
    uBit.display.scroll("D");
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_CONNECTED, onConnected);
    uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_DISCONNECTED, onDisconnected);
       
    // Note GATT table size increased from default in MicroBitConfig.h
    // #define MICROBIT_SD_GATT_TABLE_SIZE             0x500
    uart = new MicroBitUARTService(*uBit.ble, 32, 32); 
    uBit.display.scroll("UART ready");

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}
`

jamesadevine, 2016-04-18:
@bluetooth-mdw I assume these were two separate tests, and not one immediately after the other?

If it was not the latter, can you conduct the same test as two separate tests, just to scope the issue a bit.

Thanks 😄

bluetooth-mdw, 2016-04-18:
Arrrggghh. Double negatives :-)

To be clear these were two separate tests with a reset in between. So in each test I reset then connect over BLE (and you see "C" displayed)... the code then blocks in readUntil whilst I send sub-strings in discreet BLE writes from nRFMaster Controller. In test 1 when I perform the 4th write containing only the ":" end of message character, as expected I see the buffered 123456789012345678901234567890 displayed. In test 2 after a reset, same procedure just one extra character so 32 (buffer size) in total and I see no displayed text.

jamesadevine, 2016-04-18:
I know how much you do not not love them 😉 

Will look into it later today!

bluetooth-mdw, 2016-04-18:
I don't not not (not) like them if they don't (not (not)) use brackets for extra clarity though.

Thanks not not.

jamesadevine, 2016-04-19:
@bluetooth-mdw This is not not fixed by commit 0048a9ac8e4a964f5f0fdd004155c1a243274c65.

This was a simple mismatch with the buffer size provided not being the same as the actual available buffer size. The fix in that commit, is to increment the given buffer sizes by one.

Can you please retest against the latest head of master?

bluetooth-mdw, 2016-04-19:
Yep, also a job for tomorrow.

bluetooth-mdw, 2016-04-20:
My MICROBIT_BLE_EVT_CONNECTED and MICROBIT_BLE_EVT_DISCONNECTED events seem to have disappeared from ExternalEvents.h. Didn't that PR get merged? Thought it had....

My UART app uses these events. Easier to re-test with them reinstated.

jamesadevine, 2016-04-20:
Certainly are in the current master of DAL. Was using them yesterday!

Git pull?

James.

bluetooth-mdw, 2016-04-20:
Ignore me... I'd deleted that comment but you clearly saw it in your email. User error. Will test shortly.

bluetooth-mdw, 2016-04-20:
Looking good. Thanks James.

closed 2016-04-20 by bluetooth-mdw