lancaster-university/microbit-dal #126: DFU and Event Service should not be automatically included

bluetooth-mdw, 2016-04-19:
These services do not need to be explicitly instantiated by the application developer. I think this should be the way it should work. If nothing else we're stealing memory from the application. I just hit problems with a small application which uses the UART service which I believe to be memory related. I don't need DFU or Event for my application so it would make sense for them not to be there.

bluetooth-mdw, 2016-04-19:
FYI MicrobitConfig.h still offers flags to include or exclude these services but the flags no longer work:

`#ifndef MICROBIT_BLE_DFU_SERVICE
#define MICROBIT_BLE_DFU_SERVICE                0
#endif

#ifndef MICROBIT_BLE_EVENT_SERVICE
#define MICROBIT_BLE_EVENT_SERVICE              0
#endif`

jamesadevine, 2016-04-19:
Hi @bluetooth-mdw 0048a9ac8e4a964f5f0fdd004155c1a243274c65 addresses this issue. These configuration options may have been lost in the component refactor!

Could you please pull head of master, and retest? :smiley:

bluetooth-mdw, 2016-04-19:
Will do. Job for tomorrow :-)

jamesadevine, 2016-04-19:
@bluetooth-mdw you could also use the new `config.json` to test if you want! 

#122

bluetooth-mdw, 2016-04-20:
Tested using MicrobitConfig.h because I'm so old school. I'll use config.json in a year or two from now when I've got used to the idea of change ;-)

Working too btw.

closed 2016-04-20 by bluetooth-mdw