lancaster-university/microbit-dal #130: off-by-one in microbit storage

mmoskal, 2016-04-19:
```
memcpy(pair.key, key, min(sizeof(pair.key), strlen(key)));
```

https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitStorage.cpp#L218

I think this should be strlen(key)+1. Also, if strlen(key) >= sizeof(pair.key) I think the function should fail, because otherwise strcmp() further down will get into comparing random bytes in the data.

Also, this `memcpy(pair.value, data, sizeof(pair.value));` assumes 'data' is readable for at least 48 bytes. I guess if it's on the stack, near the bottom (end of memory) this could be an issue?

jamesadevine, 2016-04-20:
Good catch with the off by one, although it doesn't affect general operation. If you were to create a `ManagedString` from the key returned from the KVS, then this would be an issue. Will fix ASAP.

Not sure if the latter point applies, it should do no harm really, you would end up with garbage in flash. When reading the stored data, you would most likely `memcpy` the exact size of the struct, ignoring the garbage bytes.

jamesadevine, 2016-04-20:
@mmoskal from the `strcmp` description:

>This function starts comparing the first character of each string. If they are equal to each other, it continues with the following pairs until the characters differ or until a terminating null-character is reached.

As long as the string you give it is null terminated, it shouldn't get to the point where it spins of comparing random bytes of memory. 

If you give it a string to store thats > 16 bytes, that's user error, and you should suffer the consequences 😈 .

Happy to return a null pointer, but then it's hard to differentiate from the condition where the KVS is full!

jamesadevine, 2016-04-20:
Happy to return an error code, something like `MICROBIT_INVALID_PARAMETER` if the key is too long. Can't see any reasons as to why this would be a bad thing!

mmoskal, 2016-04-20:
I think `MICROBIT_INVALID_PARAMETER` will solve the `\0`-termination problem as well, because the strings in flash will be `\0`-terminated. Thanks!

jamesadevine, 2016-04-25:
@mmoskal patch has been applied for `MicroBitStorage` - the API has also changed to accept a size in the case where data is near the top of the stack.

closed 2016-04-25 by jamesadevine