lancaster-university/microbit-dal #132: BLE Pairing mode crashes on mbed.org based builds

finneyj, 2016-04-22:
micro:bit appears to hard fault when a BLE pairing attempt is requested... But behaves correctly for offline builds based on yotta / gcc.

bluetooth-mdw, 2017-06-04:
It does indeed. Very easy to reproduce. Happens for all possible values of MICROBIT_BLE_SECURITY_LEVEL as well.

If you reduce MICROBIT_NESTED_HEAP_SIZE to 0.50 the problem no longer occurs, albeit at a cost. I guess recent changes have pushed memory use over some threshold?

So, needs some optimisation somewhere or else it needs MICROBIT_NESTED_HEAP_SIZE  to have a new default value.

bluetooth-mdw, 2017-06-06:
See https://developer.mbed.org/users/bluetooth_mdw/code/microbit-pairing-test/