lancaster-university/microbit-dal #133: Hardware Files

mariobaldini, 2016-04-22:
Hello, 

Is the microbit hardware opensource? If yes, where could I obtain the schematics/laytouts to build one? 

I researched the web about this but couldn't figure it out. I found opensource sofware/firmware, but no mention if the hardware itself will be opensourced. 

Thank you

finneyj, 2016-04-22:
Hi @mariobaldini 

The hardware designs will be open sourced, but isn't yet. I believe the BBC are scheduled to release the remaining parts of the project in May...

mariobaldini, 2016-04-22:
Thank you @finneyj. 

I'll keep this issue open, since it is pending yet. So when they are release I update with the reference to the files. 

[]'s

finneyj, 2016-04-25:
thanks @mariobaldini

jaustin, 2017-02-01:
Hi @mariobaldini - we're still waiting for the final micro:bit schematic to be released, but in the meantime we (the Microbit Foundation) have produced a reference design for micro:bit that is completely hardware compatible but based on an nRF51 module so much easier to make yourself. 

http://tech.microbit.org/hardware/reference-design/

If you'd like to make something derived from micro:bit then this is a great place to start. I understand, though, that it's not the original schematics and those are making their way through a release process.

mariobaldini, 2021-11-25:
Thanks @jaustin and @finneyj

closed 2021-11-25 by mariobaldini