lancaster-university/microbit-dal #136: bug in message bus?

mmoskal, 2016-04-25:
I'm getting a warning from GCC and it seems spot  on:

```
/home/build/prj2/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:240:61: warning: suggest parentheses around operand of '!' or change '&' to '&&' or '!' to '~' [-Wparentheses]
         if (l->flags & MESSAGE_BUS_LISTENER_DELETING && !l->flags & MESSAGE_BUS_LISTENER_BUSY)
```

https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitMessageBus.cpp#L240

jamesadevine, 2016-04-25:
Very weird, not seeing this warning at all from GCC on my end. Definitely buggy.

dpgeorge, 2016-04-25:
I've been getting this warning ever since the code existed.  gcc version 5.3.0.

jamesadevine, 2016-04-25:
```
Scanning dependencies of target microbit-dal
[ 72%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MemberFunctionCallback.cpp.o
[ 72%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitFiber.cpp.o
[ 72%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitCompat.cpp.o
[ 73%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitDevice.cpp.o
[ 73%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitListener.cpp.o
[ 75%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitFont.cpp.o
[ 75%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitHeapAllocator.cpp.o
[ 76%] Building CXX object source/CMakeFiles/microbit-dal.dir/core/MicroBitSystemTimer.cpp.o
[ 77%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/ManagedString.cpp.o
[ 77%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/Matrix4.cpp.o
[ 78%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/MicroBitEvent.cpp.o
[ 79%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/MicroBitImage.cpp.o
[ 79%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/PacketBuffer.cpp.o
[ 80%] Building CXX object source/CMakeFiles/microbit-dal.dir/types/RefCounted.cpp.o
[ 81%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/DynamicPwm.cpp.o
[ 81%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitAccelerometer.cpp.o
[ 82%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitButton.cpp.o
[ 83%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitCompass.cpp.o
[ 83%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitCompassCalibrator.cpp.o
[ 84%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitDisplay.cpp.o
[ 84%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitI2C.cpp.o
[ 85%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitIO.cpp.o
[ 86%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitLightSensor.cpp.o
[ 86%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitMessageBus.cpp.o
[ 87%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitMultiButton.cpp.o
[ 88%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitPin.cpp.o
[ 88%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitRadio.cpp.o
[ 89%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitRadioDatagram.cpp.o
[ 90%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitRadioEvent.cpp.o
[ 90%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitSerial.cpp.o
[ 91%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitStorage.cpp.o
[ 92%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/MicroBitThermometer.cpp.o
[ 92%] Building CXX object source/CMakeFiles/microbit-dal.dir/drivers/TimedInterruptIn.cpp.o
[ 93%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitAccelerometerService.cpp.o
[ 93%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitBLEManager.cpp.o
[ 94%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitButtonService.cpp.o
[ 95%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitDFUService.cpp.o
[ 95%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitEventService.cpp.o
[ 96%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitIOPinService.cpp.o
[ 97%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitLEDService.cpp.o
[ 97%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitMagnetometerService.cpp.o
[ 98%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitTemperatureService.cpp.o
[ 99%] Building CXX object source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitUARTService.cpp.o
[ 99%] Building ASM object source/CMakeFiles/microbit-dal.dir/asm/CortexContextSwitch.s.obj
[100%] Linking CXX static library microbit-dal.a
```

Absolutely bizarre! If we had seen it - definitely would have fixed it. I'm using `gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]`

mmoskal, 2016-04-25:
I've also not seen it on my regular Windows build, but it occurs in latest Docker image for yotta, where it has:

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]
```

finneyj, 2016-04-25:
Thanks guys!

I don't see this on my yotta build either:
```
gcc-arm-none-eabi-4_9-2015q2-20150609-win32.exe
```

We should certainly disambiguate this:

```
if ((l->flags & MESSAGE_BUS_LISTENER_DELETING) && !(l->flags & MESSAGE_BUS_LISTENER_BUSY))
```

jamesadevine, 2016-04-25:
Thanks @mmoskal for bringing this up! 

Must update gcc at some point!

 cc381fe

closed 2016-04-25 by jamesadevine

finneyj, 2016-04-25:
thanks @jamesadevine