lancaster-university/microbit-dal #138: MicroBitComponent::isIdleCallbackNeeded is never called

martinwork, 2016-04-28:
Many components override isIdleCallbackNeeded() but the method is never actually called anywhere that I could find, so don't expect to turn off idleTick using isIdleCallbackNeeded!

Some components override idleTick() but not isIdleCallbackNeeded(). The base class implementation returns 0, so actually using isIdleCallbackNeeded() would break them.

Some components that implement isIdleCallbackNeeded() repeat the same check in idleTick() but do things before the check, so could also be broken by using isIdleCallbackNeeded().

finneyj, 2016-04-28:
thanks @martinwork.

isIdleCallbackNeeded is legacy code, and should be deprecated. As you say, no longer used. Old versions of the runtime had this as an optimization mechanism, but as developed continued it became apparent that the complexity wasn't worth the benefit, so is no longer used.

I'll take out all references to it.

finneyj, 2016-04-29:
Fixed in master.

closed 2016-04-29 by finneyj