lancaster-university/microbit-dal #141: serial output truncated to 20 bytes

arturl, 2016-05-06:
Calling  `uBit.serial.send` with a string longer than 20 characters only emit 20 characters.

jamesadevine, 2016-05-09:
@arturl This is by design. 

I'm unsure whether we should wrap this, or whether the users application should be responsible for wrapping this.

`uBit.serial.send` returns the number of bytes sent, if you give a string greater than the current buffer size, then it will truncate it down, and returns the number of bytes it actually sent.

Traditionally I think this is the approach most circular buffer implementations take.

However, this could be a place where our different modes could come into play... perhaps `SYNC_SLEEP` should block until **ALL** characters have been sent, not just the just the characters in the buffer.

Thoughts @arturl @finneyj ?

pelikhan, 2016-05-09:
My toughts is that we should be able to tell the runtime to send the entire buffer to avoid clients implementing their own logic to send chunks. Seems that the runtime has more knowledge/opportunities to handle this in a memory/perf efficient way.

jamesadevine, 2016-05-10:
@arturl I retract my earlier statement:

> this was by design

I believe `SYNC_SLEEP` and `SYNC_SPINWAIT` modes ***should have** completed transmission of all bytes before returning control to the user.

`ASYNC` should behave exactly as you describe.

Apologies for the confusion here, I have patched this bug and committed to master [3b587e1].

arturl, 2016-05-10:
Sounds good to me, thanks!

pelikhan, 2016-05-10:
Any chance to tag this release?

jamesadevine, 2016-05-11:
Hi @pelikhan, this is in `v2.0.0-rc3`.

closed 2016-05-13 by jamesadevine