lancaster-university/microbit-dal #144: Where is setPull?

Drmikej, 2016-05-12:
Simple question - where can I set the pull mode?
Its in the documentation but I cant find it in the code.

jamesadevine, 2016-05-12:
Hi @Drmikej,

This is likely due to your `microbit-dal` yotta module being out of date.

Can you confirm to me that in `yotta_modules/microbit-dal/module.json` is old? (the current version is `v2.0.0-rc3`)

Once you have confirmed this, you will need to do two things:

* `git pull` in the microbit-samples directory (if you have one)
  * alternately, update the microbit dependency in your root module.json
* run `yt update`

Drmikej, 2016-05-13:
The installed version on my desktop   version": "2.0.0-rc2"
and on the mbed website "version": "2.0.0-rc1",
So both are out of date. There seems to be no way to update on the mbed website and when I try to use Yotta to do it it complains that its not in the registry.

jamesadevine, 2016-05-13:
@drmikej did you follow the steps above? That should resolve the yotta version being out of date.

We push to mbed less frequently due to the frequency of commits at the moment as we are still in the release candidate phase.

Drmikej, 2016-05-13:
Update: I uninstalled the module deleted all module folders and reinstalled it from the GitHub repo.
After this the build works and setPull is ok but there is no OpenDrain 
typedef enum {
    PullNone = 0,
    PullDown = 1,
    PullUp = 3,
    PullDefault = PullUp
} PinMode;

jamesadevine, 2016-05-13:
@drmikej Thanks, the mbed documentation for [`DigitalIIn.mode()`](https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/api/DigitalIn.h#L77) is incorrect. I will update our doxygen comments.

jamesadevine, 2016-05-13:
@Drmikej Ah, the `PinMode` of course is platform dependent! Will remove the `OpenDrain` specification from the documentation.

jamesadevine, 2016-05-13:
Documentation updated: http://lancaster-university.github.io/microbit-docs/ubit/io/#setpull .

Drmikej, 2016-05-13:
Great - for the record the resistor typically 10K.

jamesadevine, 2016-05-13:
Fantastic, thanks @Drmikej, closing!

closed 2016-05-13 by jamesadevine

jamesadevine, 2016-05-13:
@Drmikej One last thing, I've update the dependencies of `microbit-samples` and `microbit-dal` to always pull the latest release, so this will hopefully avoid future versioning issues like this.

We are also planning on updating the documentation to include a "included in version" string attached to each member function!

Drmikej, 2016-05-13:
The first is a great idea.
The second not so good - I think it will become confusing. All that matters is what is supported in the current version and what is not (yet).

jamesadevine, 2016-05-16:
@Drmikej Well we were thinking of only adding the "included in version" on new member functions only.

Why would this be a bad idea?

Drmikej, 2016-05-16:
Its not a terrible idea and better than nothing but if you consider the default case - most users want and do use the latest version - most users are only interested in if the feature is implemented in the latest version or not. This means they don't care if it was in version 2,3.3 only if it is or isn't in the latest version. If you put "x is in 2.1.2 rc2 the user has to go and find out which version they have. 
This is a storm in a very small tea cup - your idea is good and better than nothing but really all that is needed is implemented Yes/No.

finneyj, 2016-05-16:
A bigger problem is usually documenting when things go away...

A simple alternative may be to have documentation track the latest tagged release (with no specific version numbers), but record the version when something is deprecated.

carlosperate, 2016-05-17:
Can the documentation be tagged and archived synchronously with the DAL? So there is no need to have a "introduced in x", just go back through older versions of documentation to see if it was there or not.

jamesadevine, 2016-05-17:
@carlosperate We did discuss this, and we concluded that it would:

1) Possibly confuse users, as browsers cache URLs and it would be possible to view different versions of the documentation without intending to.

2) Add messiness to an otherwise a consistent experience. I.e. how would you nicely expose different versions of the documentation? The npm approach of the version in the url is ok, but isn't super good in terms of usability!

Interested in hearing your thoughts on those two points @carlosperate! :smile:

Drmikej, 2016-05-17:
As your customer i.e. I read the docs and rely on them I would prefer the docs to be a singleton. I want a single authoritative document. What matters is that the docs describe the version that most people are using i.e. what you stamp as the current version. And it should be easy to match current version with current docs.
I was confused at first by what was the current version - rc1 from mbed, rc2 in my offline compiler or as I discovered rc3. I'm not even sure that calling something with rc in its name a current version is reasonable. You have people using your code you really should stamp something as release - even if your not 100% happy with it. 
Sorry to sound complaining but you need to keep in mind that your end users are already swimming in complexity and doubt and the best way to not increase it is to be bold and clear.

jamesadevine, 2016-05-17:
@Drmikej we always welcome feedback, it's the most important thing in open source projects! :smile:

I agree with the sentiment that docs should be singular and authoritative! 

We are using the terminology v2.0.0-rcX purely because we have recently radically changed from a previous iteration of the DAL, and as such users from other communities that have used the previous iteration may have comments that may further alter the way the DAL currently looks. We are looking to finalise this in the immediate future, of course!

Drmikej, 2016-05-17:
As to feedback --- have you considered a forum?
It seems odd to be having an interesting conversation in a github issue that really isn't a github issue.
There also is no micro:bit forum for users who aren't teachers - unless I've missed it.

bluetooth-mdw, 2016-05-17:
We're trying really hard to get a stackexchange forum in place. Help us by supporting the proposal:

Drmikej, 2016-05-17:
Going there right now.

whaleygeek, 2016-05-17:
The problem with singleton documentation that you describe @Drmikej is that schools will install something once at the start of term (e.g. Mu) and not have the time to re-install it several times throughout the term. So if the docs are online and always moving forward, and it's not possible to easily get the docs for the version of the thing you are using, then the docs continue to move forward and become more and more irrelevant as time goes on.

'most people are using' is a figment of everyone's imagination!

Now, downloadable, cacheable docs, for the win. I have a number of development tools and libraries that I choose myself to use a specific version of because I want to innovate, not repeatedly keep reinstalling and upgrading things. Having the library and it's associated documentation as a package together (offline cacheable) really works here.

Perhaps a way to do this (with the docs that are github hosted in particular) is to make sure that the githash (essentially a URL) of the docs that are relevant for a specific release of the code, is in some way easily related to that code. That way, if you want to, you can DownloadAsZip the docs, and DownloadAsZip the code (matching versions) and you have everything you need. Labels would probably solve the same issue (synchronised labels across the docs and code repo, so that when you look for release 2.1 of code and 2.1 of docs, you essentially get a compatible pair.

Drmikej, 2016-05-17:
Hmmmmm yes all good points.
My only issue is that if you use the mbed website to develop then the software, should in theory, be updated for you. The same is true of the majority of teachers and students using the micro:bit website. 
WIth online development 'most people are using' isn't a figment of the imagination but an enforced truth.

jamesadevine, 2016-05-25:
@Drmikej @whaleygeek @carlosperate 

With regard to the forum, I personally feel that no single entity should hold onto that much responsibility, and that the micro:bit organisation as a whole should collectively be responsible for a supportive, welcoming forum. A stack exchange site will be absolutely fantastic, but there is a need for something less formal!

The documentation generation "engine" now supports archiving the previous site. The previous versions can be accessed online using the web, or offline using git clone:

`lancaster-university.github.io/microbit-docs/archive/VERSION_NUMBER/` 

Although I expressed my opinion about the caching of urls, I feel that the url above is diverse enough for users to realise that they are on an older version of the site. It may be beneficial to display a message on the archived versions of the site to make this more transparent. 

You will begin to see archived versions as of the next increment to microbit-dal's `module.json`.

finneyj, 2016-05-25:
Awesome work - thanks @jamesadevine. Archived, version controlled pages is ideal. I was thinking it would be confusing to have versioning info on the default page, but this is very neatly addressed by tucking it away like this. nice. 

Totally agree about the forum. Something spanning all partners would be best. I mean, we could easily host one at Lancaster but it would feel very weird and unrepresentative...

carlosperate, 2016-05-25:
Sorry I was awol for a while.
I was going to suggest doc versioning that's accessible from all pages in a similar fashion to what the Python documentation has (and I think a lot of the readthedocs might have as well): https://docs.python.org/3/library/stdtypes.html
I quite like being able to quickly jump to older versions to compare differences, which is quite common when checking new releases.

That being said, I can understand why it might be a bit confusing for new users, so as long as there is a way to check older versions I am quite happy with any implementation. @jamesadevine suggestion with the explicit "archive" URL and a possible message on the page would definitely dissipate any confusion, so I'd be quite glad to see that implemented.

About the forum. I also agree that a community place would be great, and while the stack exchange might be useful to ask general questions, it's not really a site for general discussion.

Drmikej, 2016-05-25:
The problem is that the main site is for teachers and pupils and so leaves a large interested group outside. I asked the press agency for permission to post a program I'd written but was told there was no way.
Compared to the Raspberry Pi there is only a tiny micro:bit community and it isn't any any sense public. I can understand the reasons for this but it doesn't rule out a more public forum on the lines of the Pi org. This will be even more important when, and if, the micro:bit ever goes on sale to the general public.

jamesadevine, 2016-06-04:
@Drmikej @whaleygeek @carlosperate 

http://lancaster-university.github.io/microbit-docs/archive/v2.0.0-rc3

Is now live, I will eventually add an easier way to switch between versions!

carlosperate, 2016-06-06:
Glad the older versions are being archived :)
So the only current way to look at the versions is by knowing the URL? In the meantime, if the Archived folder is being manually maintained, could we have a very crude index.html page on http://lancaster-university.github.io/microbit-docs/archive/ to just displayed a link to each archived version?

finneyj, 2016-06-07:
hi @carlosperate. 

I think an index would be a really good idea. The archive is auto populated each time we push a tagged release, so I can't see why it isn't possible to create an index in this way automatically. This is @jamesadevine's baby though, and he's taking a well deserved vacation at the moment... so I'll leave this as a nice warm up exercise for when he gets back. 

Mwa-ha-ha-ha.... :-)

carlosperate, 2016-06-07:
hehe, sounds good to me, much better if the index can be auto-generated.

jamesadevine, 2016-06-21:
@carlosperate @finneyj I have planned for that, just didn't want to start implementing until I got back! :smile: