lancaster-university/microbit-dal #145: shake hard to trigger

pelikhan, 2016-05-13:
I've played rock paper scissors using the DAL shake event for a few hours and it was quite hard to trigger that gesture. Also sometimes it seems that the runtime is hung up.

Could we make the shake gesture more sensible?

finneyj, 2016-05-13:
thanks @pelikhan.

There's a damping function in the gesture recogniser that suppresses events of the same type being issued sequentially too often... most likely this damping is a little bit too strong. I'll have look.

p.s. when you say it seems the runtime is hung up, do you mean just because it was not sensitive enough with the shake events, or do you think there's something deeper?

pelikhan, 2016-05-13:
It’s hard to diagnose things with 11yo shaking around with the bits! I’ll take a deeper dive next week.

finneyj, 2016-05-13:
Indeed! thanks Peli. Let's fix up the shake event first, then see if we still have a repro. Not had any other reports of hang ups. leave it with me. I'll take a look when run out of stamina for grading exam papers.

pelikhan, 2016-05-13:
I noticed that a typical “rock paper scissors” gesture did not trigger a shake. (holding mbit in 1 hand, then 3 up-down waves + landing in other hand).

However a swirly left to right was more successfull… Anyway, it kept the kids active.

finneyj, 2016-05-14:
hi @pelikhan 

I've made some updates to make the SHAKE recogniser more sensitive, and it now raises a shake event for each shake it detects (previously it would not issue multiple back-to-back shake events). As this does seem to make more sense for general use cases.

Please see branch: bugfix-145-shake-hard-to-trigger. This looks ok in my tests, but can you try this out on your sample to see if you pick up any issues, and let me know what you find? 

@dpgeorge for visibility.

dpgeorge, 2016-05-16:
@finneyj thanks for this.  I did notice that the shake was hard to trigger so this is a welcome fix (but I did not test it myself).

finneyj, 2016-05-16:
ok - thanks @dpgeorge

finneyj, 2016-06-23:
@pelikhan @dpgeorge 

New version is now integrated into master that triggers more readily, so i'm closing this one off. If you guys see any issues, please reopen and we'll take a look.

closed 2016-06-23 by finneyj