lancaster-university/microbit-dal #146: Allow MicroBitUARTService buffer to be larger than Bluetooth MTU

bluetooth-mdw, 2016-05-17:
It would be useful if the buffer associated with a MicroBitUARTService object could be larger than the Bluetooth MTU and for data in the buffer to be automatically segmented into MTU sized chunks and transmitted as Bluetooth Indications, one at a time in response to confirmation messages coming back from the Bluetooth peer device.

jamesadevine, 2016-05-17:
@bluetooth-mdw Agreed!

I think we should reuse the Serial modes:

* `ASYNC` - Will operate as it is now.
* `SYNC_SLEEP` - Will block until all bytes have been sent.

Should have been in the original implementation, but I removed it as I wanted to keep the API consistent as there are no modes for the equivalent `read` calls. A mode for send would be a good thing!

Thanks @bluetooth-mdw .

jamesadevine, 2016-05-25:
Ammended in 	bf58aac, closing!

closed 2016-05-25 by jamesadevine