lancaster-university/microbit-dal #147: Enable all BLE services

shou7783, 2016-05-18:
Hello, 

I want my app to communicate with MicroBit via bluetooth, so I need to enable all the BLE services.
According to the document, 
```
 To enable the service, simply create an instance of this class in your program, at any time after the uBit object has been initialised
```
I try it using the following code (built by `yotta`)

```
#include "MicroBit.h"

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    new MicroBitAccelerometerService(*uBit.ble, uBit.accelerometer);
    new MicroBitButtonService(*uBit.ble);
    new MicroBitIOPinService(*uBit.ble, uBit.io);
    new MicroBitLEDService(*uBit.ble, uBit.display);
    new MicroBitMagnetometerService(*uBit.ble, uBit.compass);
    new MicroBitTemperatureService(*uBit.ble, uBit.thermometer);

    release_fiber();
}
```

After flash code to MicroBit, I can only find the accelerometer service, and there is only one characteristic found in the service.

Is there any problem or missing steps during the process?

bluetooth-mdw, 2016-05-18:
This video may help you,,,,sounds like you need to tweak memory a bit:

http://bluetooth-mdw.blogspot.co.uk/p/bbc-microbit.html#mbed_ble

finneyj, 2016-05-18:
Indeed - thanks @bluetooth-mdw.

@shou7783 The bluetooth stack reserves a big chunk of memory to hold BLE services (up to  around 1.8K in fact). As we're very short on memory for the microbit, we reclaim some of this for most builds - as the memeory is typically better spent supporting kids code. So, just a small tweak of configuration required to give the memory back to the BLE stack.

See the blue note on the BLE services pages for more info:
http://lancaster-university.github.io/microbit-docs/ble/accelerometer-service/

Also, information here on how to set configuration options (it's subtly different on mbed.org to offline yotta builds):
http://lancaster-university.github.io/microbit-docs/advanced/#compile-time-configuration-options

jamesadevine, 2016-05-18:
Also, the microbit-samples project contains a config.json, which will override config options set in the dal.

I'm due to remove this at a later date, as it's misleading given the docs detail a different way by default!

James.

http://lancaster-university.github.io/microbit-docs/ble/accelerometer-service/

http://lancaster-university.github.io/microbit-docs/advanced/#compile-time-configuration-options

finneyj, 2016-05-18:
thanks @jamesadevine. i think that's a very good idea. Having the yotta options as a way to configure the runtime is a good thing, but having defaults set in multiple places could easily lead to confusion.

Perhaps a documenting the options in a comment block in the yotta configuration would be a good compromise.

shou7783, 2016-05-24:
Thanks all!! I can find all the services and characteristics after modifying all the `MICROBIT_SD_GATT_TABLE_SIZE` variable to `0x700`.

Now, I enable the `Temperature`, `Accelerometer`, `Magnetometer`, `Button` notifications, but I can only receive the `Button` notifications. And I try to read the value of these characteristics, I get `0` value except `Button`. 

After a few tries, I found that if I add the following code, I can receive accelerometer notification.
```
while(1) {
        uBit.accelerometer.getX();
}
```

Is there any missing code if I want to receive all the notifications & values ?

finneyj, 2016-05-24:
@shou7783 

that's strange - you shouldn't need to do that. It sound a bit like the device drivers didn't wake from energy saving mode...

Can you let us know which version of the code you're running? This will be reported in your BLE Device Information service, and also in the module.json file?

Also, maybe try your code again against the latest version of the microbit and microbit-dal repos? There was a 'feature' a little like this a month or so ago that was fixed.

finneyj, 2016-06-23:
@shou7783 

Just checking if this was resolved by the above suggestions? If so, is this issue safe to close?

bluetooth-mdw, 2016-06-23:
Pretty sure I've seen the accelerometer issues as well btw...

shou7783, 2016-06-24:
I just added while loop in the end of code for the notification issue, and it works for me.