lancaster-university/microbit-dal #149: Question: Serial Access

Steve136, 2016-05-25:
Hello, 

I am looking at integrating an ESP8266 module via Serial. I have made the connection and can send/receive data perfectly fine. 

Is it possible to use the PC Serial connection on USBRX, USBTX as well as Serial on two pins? 

When I try this, either one or the other works... but not both? 

Thanks.

jamesadevine, 2016-05-25:
@Steve136 you can use [redirect](http://lancaster-university.github.io/microbit-docs/ubit/serial/#redirect) to redirect the output of a serial instance, or you simply can create another instance of `MicroBitSerial` with your pins of choice.

Steve136, 2016-05-25:
I have created an instance of MicroBitSerial with USBRX and USBTX - Works fine on it's own. 

If I create another instance of it, or use just Serial with PIN1 and PIN2 for RX/TX for my module then the PC USB serial does not work.

If I comment out one of them the other works, but not both at the same time. Essentially I want to print the output to the PC USB serial from my ESP8266 serial.

remay, 2016-05-25:
If I remember correctly the SoC only has a single UART.   So not possible to use 2 serial connections simultaneously.  But, depending on what you are trying to do you might be able to use redirect() to move the pins back and forth between the USB and your external device, using each in turn?

finneyj, 2016-05-26:
Yes indeed - only one UART, so whenever you create another serial instance, you'll just replace the previous one I'm afraid.

@remay's suggestion is a good one. As @jamesadevine says, the redirect() method should help you here. You'd only need to move the TX pin of course, so should never lose any data from your ESP8266...

Steve136, 2016-05-27:
Thank you! That sounds exactly what I was looking for. I may be doing something wrong though?

The code below prints "Microbit Started.." fine but the program appears to halt on the SC.Send("AT") and not go any further? It displays the "OK" message but It never shows the "Redir". The redirect does return MICROBIT_OK.

Am I using this correctly? 

`#include "MicroBit.h"

MicroBit uBit;
MicroBitSerial SC(USBTX, USBRX);

int main() {
   uBit.init();
   uBit.sleep(5000);
   
    while(1) {
      SC.send("Microbit Started... \r\n");

      int ESPModule = SC.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
      if (ESPModule == MICROBIT_OK) {
        uBit.display.scroll("OK", 100);
        SC.send("AT");
      }
      
      uBit.display.scroll("Redir", 100);
      uBit.sleep(1000);
      
      int USBSerial = SC.redirect(USBTX, USBRX);
      if (USBSerial == MICROBIT_OK) {
          uBit.display.scroll("OK", 100);
          SC.send("Hello PC!\n");
      }
      
      SC.send("Finished.");
      uBit.sleep(10000);
    }
}`

Thank you

jamesadevine, 2016-05-27:
Hi Steve,

I'm on the move at the moment, but there is already a serial object attached to uBit (uBit.serial), so there could be some form of interaction there!

James.

Steve136, 2016-05-27:
Thanks for that! Makes sense.  I've switched to the uBit serial - I still get the same though. Outputs fine first and appears to hang on the send() and nothing after that. (It could be just me?) code below.

`#include "MicroBit.h"
MicroBit uBit;

int main() {
   uBit.init();
   uBit.sleep(5000);
   
    while(1) {
      uBit.serial.send("Microbit Started... \r\n");

      int ESPModule = uBit.serial.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
      if (ESPModule == 0) {
        uBit.display.scroll("OK", 100);
        uBit.serial.send("AT");
        uBit.sleep(5000);
      }
      
      uBit.sleep(1000);
      
      int USBSerial = uBit.serial.redirect(USBTX, USBRX);
      if (USBSerial == MICROBIT_OK) {
          uBit.display.scroll("OK", 100);
          uBit.serial.send("Hello PC!\n");
      } else {
          uBit.display.scroll("BAD.");
      }
      
      uBit.serial.send("Finished.");
      uBit.sleep(10000);
    }
}`

finneyj, 2016-05-27:
nope - just tried this quickly. Also acts the same with uBit.serial.send() / uBit.serial.redirect().

Ingerestingly though, the problem doesn't surface if you use uBit.serial.printf() instead of uBit.serial.send(). The difference being that printf is currently a spinwait.... looks like a bug.

finneyj, 2016-05-27:
sorry @Steve136 - crossed posts there! Agree with you.

Steve136, 2016-05-27:
@finneyj No probs - I can also confirm printf() works perfectly! Thank you.

finneyj, 2016-05-27:
Cool. Suggest you run with printf() for now. 

Thanks for the report @Steve136 suspect some incorrect configured state after a redirect() that's affecting the locking... send blocks on the scheduler rather than spinwaiting, which is a bit more power efficient and better for concurrency.

jamesadevine, 2016-05-27:
@Steve136 @finneyj Interesting, I will investigate!

jamesadevine, 2016-05-27:
Okay, I've tracked it down to [`serial_free()`](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitSerial.cpp#L838). It seems to be removing some state without tearing down the interrupts that rely on it. 

@Steve136 , could you verify that by removing `serial_free()` from your local MicroBitSerial.cpp your program above works?

Steve136, 2016-05-27:
@jamesadevine That seems to work for me. Commented out serial_free(&_serial); in MicroBitSerial.cpp made the original program run as expected with the send() and redirect() function.

Added it back in to MicroBitSerial and recompiled, it broke it again, so it's definitely that!

Thank you!

finneyj, 2016-05-27:
sounds good - thanks guys.

@jamesadevine, can you please merge this fix into master when you get a mo, then close this issue off.

jamesadevine, 2016-05-27:
Thanks @Steve136.

@finneyj a patch is now in master [179b32f] and will be available as of the next tagged release.

closed 2016-05-27 by jamesadevine