lancaster-university/microbit-dal #150: Incorrect sequencing in Radio IRQ handler?

whs-che, 2016-05-30:
I wrote a simple checksum algorithm at my application level when I suspected packet corruption. To my surprise it did actually trap errors. I was rather hoping the lower level routines would discard packets with a bad CRC check. I then thought that maybe this was deliberate for education purposes and that the rssi value (only updated if no CRC error) was intended as an indication of the failure (value 0 maybe?). This does not work for two reasons: firstly the rssi is never explicitly set under a CRC fault and would rely on a memory allocation doing so and secondly the rssi value appears to be stored in the context of of the new receive buffer and not the one which has just been sent to the queue.

Without knowing the original intention I have fixed the problem by calling the following  two lines of code only if the CRC is okay and not at the start of the routine as in the current version:

 // Move on to the next buffer, if possible.
            MicroBitRadio::instance->queueRxBuf();
            NRF_RADIO->PACKETPTR = (uint32_t) MicroBitRadio::instance->getRxBuf();

Note that queueRxBuf() internally takes the last rssi value and queues before it has chance to be updated if called at the start of the ISR.

There's a distinct possibility that I will cause unintended side effects with this mod but all I know for now is that it's working perfectly. 

Hope this all makes sense and is of some help.

finneyj, 2016-05-31:
@whs-che 

Many thanks for feeding this back - this is indeed a missed validation check, and this looks like a good fix. Would you like to present it as a pull request so we can merge this in wit your attribution?

finneyj, 2016-06-03:
@whs-che 

I've tested and merged your bugfix documented here. Thanks for the contribution!
https://github.com/lancaster-university/microbit-dal/commit/07b6f7f6b65e9b566c86abe7c471ad7dfb41851e

closed 2016-06-03 by finneyj