lancaster-university/microbit-dal #151: Serial triggers microbit hang, even when UART is not attached!

cefn, 2016-05-30:
Thanks to everyone for their efforts on Microbit, and especially for @jamesadevine for his helpful suggestions so far (moving our discussion elsewhere to this issue).

I am experiencing an apparent hang when trying to load up the Serial connection on a Microbit to connect to an MH-Z16 sensor with a 3.3V Uart as documented at http://www.seeedstudio.com/wiki/images/c/ca/MH-Z16_CO2_datasheet_EN.pdf

When the following code which arose from a discussion with @jamesadevine is compiled on the mbed compiler, and downloaded to a Microbit, nothing is shown on screen at all, whilst I believe at least the 'C' should be showing. This suggests that there's a hang, despite the simplicity of the example.

This is assuming that scrollAsync should keep running even when the serial is waiting for data.

```
#include "MicroBit.h"

MicroBit uBit;

const char commandbytes[] =  {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};

ManagedString command(commandbytes, 9);

int main()
{
    uBit.init();
    uBit.serial.baud(9600);
    uBit.serial.redirect(P0_14, P0_15);

    while(1)
    {
        uBit.display.scrollAsync("C"); //command
        uBit.serial.send(command);

        uBit.display.scrollAsync("W"); //wait
        ManagedString response = uBit.serial.read(9);

        uBit.display.scrollAsync("S"); //show
        char header = response.charAt(1);
        char highppm = response.charAt(2);
        char lowppm = response.charAt(3);
        if(header == 0x86)
            uBit.display.scroll("D"); //indicates success
        else
            uBit.display.scroll("?"); //indicates failure
    }
}
```

However, even sensor failure doesn't explain why the Microbit can't show the async logging inlined in the code even with nothing attached at all, which suggests it's a hang arising from the Serial initialisation, not that it's waiting for data.

However, I'd expect the circuit to report data when the sensor is wired in too. I ran the sensor for a day or so against a [Shrimp](http://shrimping.it) based circuit, (including 3.3V to 5v logic level shifter). In this setup I successfully used the [NDIR Arduino library](https://github.com/SandboxElectronics/NDIR) to dump the CO2 PPM many thousands of times over a day when running it in. 

Then I migrated to a Microbit implementation based on Micropython (where serial redirection and bytewise reading are both supported, unlike the other higher-environments) and I succeeded at reading data from the sensor for several cycles and printed it out on the display before it hung, as described at https://github.com/bbcmicrobit/micropython/issues/286 

After it's hung, this seems to be a semi-permanent state, reset by black magic in true Heisenbug fashion. Sometimes replugging it works for a few cycles, sometimes powering up the sensor after the microbit works for a few cycles, sometimes only reprogramming the microbit helps, and then for hours nothing helps at all. I've tried this over three separate microbits from two different releases, including the pupil release which is the main build I've been testing.

It's always feasible that the sensor is somehow broken since yesterday, but it's hard to square with it working fine for a day on a Shrimp build, and working intermittently with occasional hangs in a Micropython-driven Microbit circuit. It's also orthogonal, since the code above seems to hang even on an unwired Microbit.

cefn, 2016-05-30:
The following example might be easier to debug...

```cpp
#include "MicroBit.h"

MicroBit uBit;

const char commandbytes[] =  {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};

ManagedString command(commandbytes, 9);

int main()
{
    uBit.display.scroll("I"); //initialise
    uBit.init();

    uBit.display.scroll("B"); //baudrate
    uBit.serial.baud(9600);

    uBit.display.scroll("R"); //redirect
    uBit.serial.redirect(P0_0, P0_1);

    while(1)
    {
        uBit.display.scroll("L"); //loop

        uBit.display.scrollAsync("C"); //command
        uBit.serial.send(command);

        uBit.display.scrollAsync("W"); //wait

        ManagedString response = uBit.serial.read(9);

        uBit.display.scrollAsync("S"); //show

        char header = response.charAt(1);
        char highppm = response.charAt(2);
        char lowppm = response.charAt(3);

        if(header == 0x86)
            uBit.display.scroll("D");
        else
            uBit.display.scroll("?");
    }
}
```

On my hardware, it scrolls the following...

```I   B   R   L```

...indicating that the problem arises at the moment the command is invoked, (blocking the async scrolling of the letter 'C' with a hang).

Note I switched from using P0_14 and P0_15 as they caused a different problem, with a sequence that looked like...

```I   B   R   L  [KLINGON]```
 
...where the Klingon lettering isn't discernable, but could be some corruption of a display, or potentially a deliberately placed fail-whale error logo. Using pins 0 and 1 avoids this problem.

jamesadevine, 2016-05-30:
@cefn running your code, there are a few things here...

Your `ManagedString` instance (command) is actually empty as there is some sanity checking in the constructor where if the buffer contains bad characters, such as a `\0`, before your given length (your does) it will return an empty string. 

More simply you could directly send the buffer over serial:

```
uBit.serial.send((uint8_t *)commandbytes, 9);
```

Furthermore, there is no error checking on the send command for a length of zero, I will add this in in the morning, it enters an infinite loop. 

You will also need to use the `serial.read(buffer, bufferLen)` as I note that your external component will return data with terminating characters:

```
char recv[9] = { 0 };

uBit.serial.read((uint8_t *)recv, 9);
```

`ManagedString` is really intended for more simplistic applications, @finneyj we should definitely implement a more general buffer  (#112) to handle these cases.

Additionally, don't directly use P0_0, this is the literal pin on the NRF chip, which is very different to a pin on the edge connector. You can use the pin names listed towards the end of [this file](https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_MICROBIT/PinNames.h) directly, or alternately use the far clearer pin mappings in [MicroBitPin.h](https://github.com/lancaster-university/microbit-dal/blob/master/inc/drivers/MicroBitPin.h#L42).

You will need to apply the patch [here](https://github.com/lancaster-university/microbit-dal/issues/149) for `serial.redirect` to work as expected.

Apologies i didn't catch this earlier in our email chain, I was out and about 😛

cefn, 2016-05-31:
Wow, thanks for the late night response above and beyond the call of duty, @jamesadevine ! I've updated the code with the modifications you mention, as well as removing the 'free' line 824 from microbit/microbit-dal/source/drivers/MicrobitSerial.cpp within the Mbed compiler tree for this program.

The updated code is as below, and is able to complete a loop and maintain liveness, thankyou. 

However, at present while the read operation returns a count of 9, it doesn't appear to actually populate the respondbytes array, and the values remain at 0xFF. If I switch the default contents of the array to zeroes, they all remain at 0x00. So with the code below the microbit currently scrolls...

```S 9 255 255 255```

I've just verified the sensor with the Shrimp test circuit and re-wired it, proving the Shrimp (with logic level converter from 3.3V to 5V), can successfully read the bytes returned by the sensor, and do the ppm calculation, so it must be sending the bytes. Also I guess the microbit UART must be picking up edges, but the values aren't getting through.

```cpp
#include "MicroBit.h"

MicroBit uBit;

const char commandbytes[] =  {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};
const char respondbytes[] =  {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

int main()
{
    uBit.init();
    uBit.serial.baud(9600);
    uBit.serial.redirect(MICROBIT_PIN_P0, MICROBIT_PIN_P1);

    while(1)
    {
        uBit.serial.send((uint8_t *)commandbytes, 9);     
           
        int readCount = uBit.serial.read((uint8_t *)respondbytes, 9);

        char header =   respondbytes[1];
        char highppm =  respondbytes[2];
        char lowppm =   respondbytes[3];

        uBit.display.scroll("S");
        uBit.display.scroll(ManagedString((int)readCount));
        uBit.display.scroll(ManagedString((int)header));
        uBit.display.scroll(ManagedString((int)highppm));
        uBit.display.scroll(ManagedString((int)lowppm));
    }
}
```

cefn, 2016-05-31:
I have successfully got a version to work! It's reading CO2 PPM and ready to incorporate into the prototype. It's so far been running for a bunch of time and delivering fairly predictable results - if you breathe on it, it notices!

The main difference is not relying on any buffer-oriented population, which I guess borks on all the 0x00 special characters somewhere in the serial stack. Instead I'm manually enumerating the buffer characters. Not sure why that's needed, given the low level buffer copy operations I thought we were using even in the last version.

One remaining question. I was aiming for this to be the basis of a tutorial-type presentation with code that pupil learners could replicate from microbit.co.uk about 5 days from now. Is that in any way feasible? 

Is there a route by which I can get an updated serial stack available from any of the accessible scripting languages within that timescale. If not next week, do you know when any changes to the DAL are likely to drop downstream, and where (which languages) I can benefit from them, or any other way I can make it possible for learners in general to use this sensor?

Here's the working version!

```cpp
#include "MicroBit.h"

MicroBit uBit;

char commandbytes[] =  {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};
char respondbytes[] =  {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

int main()
{
    uBit.init();
    uBit.serial.baud(9600);
    uBit.serial.redirect(MICROBIT_PIN_P14, MICROBIT_PIN_P15);

    while(1)
    {
        uBit.serial.send((uint8_t *)commandbytes, 9);     
           
        respondbytes[0]=uBit.serial.read();
        respondbytes[1]=uBit.serial.read();
        respondbytes[2]=uBit.serial.read();
        respondbytes[3]=uBit.serial.read();
        respondbytes[4]=uBit.serial.read();
        respondbytes[5]=uBit.serial.read();
        respondbytes[6]=uBit.serial.read();
        respondbytes[7]=uBit.serial.read();
        respondbytes[8]=uBit.serial.read();
        
        char header =   respondbytes[1];
        char highppm =  respondbytes[2];
        char lowppm =   respondbytes[3];
        
        if(header==0x86){
            int ppm = ((int)highppm) * 256 + lowppm;
            uBit.display.scroll(ppm);            
        }
        else{
            uBit.display.scroll("?");            
        }
    }
}
```

jamesadevine, 2016-05-31:
@cefn your previous version didn't work because you declared it const

> const char respondbytes[] =  {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

Hmmm all great questions, and ones that should be answered later on this morning :wink:

cefn, 2016-05-31:
Teehee, that's pretty stupid of me. That copy-paste error is the 2 in the morning coding problem right there! Now it's scrolling on my desk I can finally relax.

jamesadevine, 2016-05-31:
@cefn classic 2am coding :wink:

@pelikhan is from Microsoft and he maintains http://m.pxt.io, it is likely that pxt could be updated in that timescale, but i'll leave that to Peli to confirm!

cefn, 2016-05-31:
Problem with current expressivity of http://m.pxt.io is that it only handles readline (newline terminated) so assumes you have control of both sides of the protocol, or that it happens to be newline terminated. I can't control how this sensor sends its data.

It's for that reason I wrote off using it in the first place, though it seems almost the best programming experience out there of the ones that I've seen. [It's funny to be revisiting this experience, as I was the instigator, architect and project lead of BT Rules... https://vimeo.com/4665910 see also unpublished from 2008 http://cefn.com/blog/files/bt_rules_paper.pdf - them sitting on this project is one of the reasons I left BT Research]

At the very least I'd need a 'readByte' which returns a number between 0-255 and equivalent 'writeByte' which doesn't serialise the number, but passes it verbatim to the UART. @pelikhan any chance these byte-oriented operations might be added to the Serial capabilities in http://m.pxt.io and that @jamesadevine's recent patch would make it to the pxt.io cloud builder in time to document for the 6th June? 

The reason timings are so important; it's in support of a Live broadcast demo on the 6th June and expected to be presented as a linked 'home build' for educators and learners to do sampling of atmospheric sensors, so we'd need to document it in a way which they could follow (not MBed Compiler).

dpgeorge, 2016-05-31:
@cefn can you confirm that with the latest working DAL version of your program that you need to separate the UART reading into 9 individual reads of 1 byte each?  And that if you try to read 9 at once it doesn't work?

cefn, 2016-05-31:
Having to do bytewise was an issue with my implementation declaring the result array as const as per https://github.com/lancaster-university/microbit-dal/issues/151#issuecomment-222569201 

This was originally a copy-paste error from the line above, but even when I noticed my bug I was expecting it would be interpreted as a const char pointer, not a pointer to const chars - this is my bad C++. The functioning code tested a couple minutes ago now looks like...

```cpp
#include "MicroBit.h"

MicroBit uBit;

char commandbytes[] =  {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};
char respondbytes[] =  {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

int main()
{
    uBit.init();
    uBit.serial.baud(9600);
    uBit.serial.redirect(MICROBIT_PIN_P14, MICROBIT_PIN_P15);

    while(1)
    {
        uBit.serial.send((uint8_t *)commandbytes, 9);     

        uBit.serial.read((uint8_t *)respondbytes, 9);
        
        char header =   respondbytes[1];
        char highppm =  respondbytes[2];
        char lowppm =   respondbytes[3];
        
        if(header==0x86){
            int ppm = ((int)highppm) * 256 + lowppm;
            uBit.display.scroll(ppm);            
        }
        else{
            uBit.display.scroll("?");            
        }
    }
}
```

My stupid 2am bug was a separate issue from the DAL issue, in the way that ManagedString handles buffers containing nulls (even when the length of the buffer is explicit), which had a knock-on effect when using the standard MicroBitSerial API to effectively pass zero-length strings. 

This is additionally separate from the way that the Serial stack responds to the mistakenly-created zero-length strings (infinite loop). 

However, it's a 'cascade failure' from having/choosing to operate with byte sequences as strings, even when some byte sequences are not strings (e.g. they contain null characters, and are not terminated by them).

However, the various issues emerging at different levels of the stack to do with satisfying this sensor's serial protocol (with frames including null values not as terminators) suggested to me how the same issue might be behind the hang I experience when sending and receiving these bytes in micropython. 

I believe it's an independent implementation so wouldn't be directly affected by any of the above cases. I would expect the micropython byte-sequence-oriented API to faithfully send all the bytes passed to it, even if some of them were nulls, but I wondered if there's an assumption built in somewhere which triggers a related problem, given how much of these byte array frames are nulls.

pelikhan, 2016-05-31:
@jamesdevine is there a tagged release?
@cefn can you open an issue in the PXT-microbit repo? Should be doable this week.

jamesadevine, 2016-05-31:
@pelikhan Not yet - want to get a couple of things in.

cefn, 2016-06-01:
In case you were wondering what the project was, here's a preview...
https://youtu.be/BLcmQM1_zHo

The home build has been shelved until we can source the cheaper version of the sensor, though that's here in a few days. 

That means there's less priority on establishing this route to wire and read gas sensors for learners in support of a Live Lesson. Instead they've gone with simpler 'environmental' sensors to illustrate the interplanetary theme, though the atmospheric sensor is the flagship example for STEM purposes given that's how we really detect for little green men (though usually it's with methane).

finneyj, 2016-06-01:
Cool. Care with kids and methane sensors. I think I know precisely what they'll start sensing and how they'll test it both at home and in the classroom! :-)

cefn, 2016-06-01:
That's how we started out. Opted for CO2 to minimise sniggers, and save the presenters having to generate methane on air somehow.

jamesadevine, 2016-06-04:
@pelikhan tagged release now available:

`v2.0.0-rc4`

@cefn would you consider this particular issue resolved now?

finneyj, 2016-06-23:
@cefn 

ping - is it safe to close this one now?

cefn, 2016-06-23:
Sorry, yes I think the fundamental issue (which consistently caused a hang) has now been identified and resolved, with only https://github.com/lancaster-university/microbit-dal/issues/152 possibly remaining and different in nature.

closed 2016-06-23 by cefn