lancaster-university/microbit-dal #155: Undefined reference to 'MicroBit::MicroBit()'

Austin1990, 2016-06-03:
#include "MicroBit.h"
MicroBit uBit;
int main()
{
    uBit.init();
    uBit.display.scroll("HELLO WORLD!");
    release_fiber();
}

Using Yotta build this I get the error:

[1/1] Linking CXX executable source/microbit
FAILED: cmd.exe /C "cd . && C:\yotta\gcc\bin\arm-none-eabi-g++.exe   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -TC:/Users/a_bowman/microbit/yotta_targets/bbc-microbit-classic-gcc/CMake/../ld/NRF51822.ld -Wl,-Map,source/microbit.map -Wl,--start-group source/CMakeFiles/microbit.dir/C_/Users/a_bowman/microbit/source/main.cpp.o  ym/microbit-dal/source/microbit-dal.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/nrf51-sdk/source/nrf51-sdk.a ym/mbed-classic/existing/mbed-classic.a -lnosys  -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/microbit && cmd.exe /C "cd /D C:/Users/a_bowman/microbit/build/bbc-microbit-classic-gcc/source && arm-none-eabi-objcopy -O ihex microbit microbit.hex && srec_cat C:/Users/a_bowman/microbit/yotta_targets/bbc-microbit-classic-gcc/CMake/../bootloader/BLE_BOOTLOADER_RESERVED.hex -intel C:/Users/a_bowman/microbit/yotta_targets/bbc-microbit-classic-gcc/CMake/../softdevice/s110_nrf51822_8.0.0_softdevice.hex -intel microbit.hex -intel -o microbit-combined.hex -intel --line-length=44 && cd /D C:/Users/a_bowman/microbit/build/bbc-microbit-classic-gcc/source && C:/yotta/gcc/bin/arm-none-eabi-objcopy.exe -O binary microbit microbit.bin""
source/CMakeFiles/microbit.dir/C_/Users/a_bowman/microbit/source/main.cpp.o: In function `main':

C:/Users/a_bowman/microbit/source/main.cpp:7: undefined reference to `MicroBit::init()'

source/CMakeFiles/microbit.dir/C_/Users/a_bowman/microbit/source/main.cpp.o: In function `__static_initialization_and_destruction_0':

C:/Users/a_bowman/microbit/source/main.cpp:3: undefined reference to `MicroBit::MicroBit()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

however, using this compiles & links ok:

#include "MicroBit.h"
MicroBitDisplay display;
int main()
{
    display.scroll("HELLO WORLD!");
    release_fiber();
}

what is wrong

finneyj, 2016-06-03:
hi @Austin1990 

Looks like you're missing the microbit module... Suggest you pull the "microbit-samples" repo from this site, which shows how this all fits together. There's some getting started instructions on our docs:
http://lancaster-university.github.io/microbit-docs/offline-toolchains/

The good news is that if your second example works, then your build environment looks good. :-)

Austin1990, 2016-06-06:
Hi @finneyj 

I went back to 1st principles, deleted what I had and re-generated the project from scratch and it now works correctly.

yotta init
yotta target bbc-microbit-classic-gcc
yotta install lancaster-university/microbit
...add source file
yotta build

Thanks.

closed 2016-06-06 by Austin1990