lancaster-university/microbit-dal #156: Rollup optional pull specification into getDigitalValue

cefn, 2016-06-03:
This is a change request following some consensus on medium-term workarounds for the surprising pull-down defaults of digitalWrite (surprising for experienced makers anyway). 

This relates to the discussion at... https://github.com/bbcmicrobit/micropython/issues/288 

The request is to provide a signature for getDigitalValue which promotes visibility of software-controllable pull behaviour and makes the existing default pull-setting visible and more accessible, as well as seamlessly allowing getDigitalValue calls to be opinionated about pull, and encouraging the downstream languages to extend their own mappings of the getDigitalValue call to include pull specification.

THE PROPOSAL 

They should be set by...

* calling getDigitalValue with an explicit pull argument (this is the extra requested signature, of which the below is a special case of resorting to a default value for this argument)
* calling getDigitalValue with no explicit pull at which point they are implicitly set to e.g.  MICROBIT_PULL_DEFAULT which is currently pull-down, though there should perhaps be a separate discussion about this. If previously set, getDigitalValue without an explicit pull will not overwrite the value. 
* explicitly pulling a pin (but not as part of a getDigitalValue call).

An alternative to avoid the potential for getDigitalValue() to be mistakenly substituted for setDigitalValue() which would end up with the same argument signature and a single character difference, is to provide getPulledValue() but this does not have the benefit of encouraging downstream languages to track an existing API they are bound to, and they may remain oblivious to the need to expose this key operation.

finneyj, 2016-06-03:
Thanks @cefn. Agree with all that, except the last point. getDigitalValue() is the known name - I think we should stuck with it.

I'll draft up a pull request (pun intended!) for review.

finneyj, 2016-06-03:
https://github.com/lancaster-university/microbit-dal/pull/158

finneyj, 2016-06-03:
@cefn @dpgeorge @pelikhan 

FYI, as requested. https://github.com/lancaster-university/microbit-dal/pull/158