lancaster-university/microbit-dal #161: IO eventOn Question?

Steve136, 2016-06-08:
Hello, 
I am wanting to measure the amount of time a pin receives a HIGH value. From looking at the runtime docs I can do this using the eventOn and using the Pulse? (or Rise/Fall)

I am trying to put together and example of this (see code below) but it never triggers the event..

`#include "MicroBit.h"
MicroBit uBit;
MicroBitMessageBus bus; 

void onPulse(MicroBitEvent evt) { 
    uBit.serial.send("\nPulsed!");
    uBit.serial.printf("\nPulse Duration: %i", evt.timestamp);
 } 

int main() {
   uBit.init();
   uBit.serial.send("\nStarting up...");
   
   uBit.io.P0.eventOn(MICROBIT_PIN_EVENT_ON_PULSE);
   bus.listen(MICROBIT_ID_IO_P0, MICROBIT_PIN_EVENT_ON_PULSE, onPulse, MESSAGE_BUS_LISTENER_IMMEDIATE); 

    while(1) {
        uBit.sleep(500);
        uBit.serial.send(".");
    }
}`

Whilst that is running, I am connecting a separate 3.3v (from a breadboard) to P0. 

This should trigger the event? Have I understood the documentation correctly, or just doing something wrong?

Thanks!

finneyj, 2016-06-08:
hi @Steve136, this looks pretty close! 

I think it's one of two simple things:

Firstly, you're creating your own MessageBus - there's actuall a default one as part of the uBit object already, so the chances are you're listening for events on one bus, but the events you want are being sent on the other. :-)  

Secondly, the MESSAGE_BUS_LISTENER_IMMEDIATE flag may not be what you want here. This tells the MessageBus that you'd like to process your event within the context of the code raising the event (so you get notified really quickly). In this case, that will be interrupt context, which we try to avoid where possible. Depending on the frequency of your pulses, you would likely be better off with just the default flags as used in the example below.

I just tried this against the master version of microbit-dal and it seems ok:

```
#include "MicroBit.h"

MicroBit uBit;

void onPulse(MicroBitEvent evt) { 
    uBit.serial.send("\nPulsed!");
    uBit.serial.printf("\nPulse Duration: %i", evt.timestamp);
 } 

int main() {
   uBit.init();
   uBit.serial.send("\nStarting up...");

   uBit.io.P0.eventOn(MICROBIT_PIN_EVENT_ON_PULSE);
   uBit.messageBus.listen(MICROBIT_ID_IO_P0, MICROBIT_PIN_EVT_PULSE_HI, onPulse); 

    while(1) {
        uBit.sleep(500);
        uBit.serial.send(".");
    }
}
```

p.s. I think this might have come from our docs being a bit too heavily structured away from people using "uBit", and is probably something we should change...

Steve136, 2016-06-08:
Thanks @finneyj that makes perfect sense.

With the code above I am seeing some events getting triggered now. For the original example code, I copied what was in the docs as a starting point, I haven't worked with the events or mbed before.

I've just tested the above and now connected up my Ultrasonic sensor and is working perfectly! 

Thank you.

finneyj, 2016-06-08:
Awesome! Thanks for feeding back - assume that's an SRF04 variant? I wanted to test one of those when we wrote these events, but I only had a 5V SRF04 to hand and was away form my workshop at the time so had to trust my oscilloscope. :-)

p.s. i note @pelikhan has all this functionality mapped into PXT now... so if you're quick, you might get to be the first person to write a third party device driver in pure Blocks for PXT. :-)

Steve136, 2016-06-08:
It's an HC-SR04 (also 5v) I am using a breadboard power supply for it currently, now working perfectly though.

Interesting idea with the Blocks, I had a quick attempt just now: https://m.pxt.io/fzdzvv - I'll see if it works properly tomorrow when I am reunited with my Micro:Bit.
 
I guess this can be closed now as it wasn't really an issue. Is there any sort of forum for the Micro:Bit?

finneyj, 2016-06-23:
hi @Steve136,

thanks for feeding back - sorry for slow response (not had much time to work on micro:bit recently). Your blocks driver looks great!

Agree we really need a good, friendly, open forum for micro:bit. There are ongoing efforts to get a stackexchange site off the ground, but even then, it feels to me like we need something more accessible, where people can share all their goodies...

closed 2016-07-15 by jamesadevine