lancaster-university/microbit-dal #162: Eddystone and microbit

danjenkins, 2016-06-09:
So I'm curious how to broadcast an eddystone URL from the micro:bit - as shown here on twitter using the runtime

https://twitter.com/microbitruntime/status/695632372746842112

I've seen its available in mbed further up the chain (I think); is it available in microbit-dal yet?

finneyj, 2016-06-09:
hi @danjenkins 

I know @jamesadevine was looking into this (but is on vacation at the moment). Looks like he has a branch here though: https://github.com/lancaster-university/microbit-dal/tree/beacon-services

Strange he didn't finish it off - looks close. You could maybe take a look? Looks a few months old now - may useful to rebase this against the current master revision.

danjenkins, 2016-06-09:
Thanks @finneyj ! I'll take a look at that!

finneyj, 2016-06-23:
no worries @danjenkins. I'm sure James will be nibbling away at integrating this when he has a moment, so I'll close this one off if that's ok.

closed 2016-06-23 by finneyj