lancaster-university/microbit-dal #163: getPull() or extra pinMode enum needed?

cefn, 2016-06-09:
Given the knock-on effects of the de-facto decision in https://github.com/lancaster-university/microbit-dal/issues/157 , seems like we need either
* a getPull() function
* a pinMode enum entry PullDon'tMessWithWhatIt'sAlreadySetTo 
...so that getDigitalValue can be prevented from having unexpected side effects.

This will make it feasible to call upon getDigitalValue without having a strong coupling with either the state of the DAL's pull interventions, or the application-designers historical first digitalWrite call, both of which may be impossible to know.

An example motivation is for code reuse where getDigitalValue is used, e.g. for displaying platform state, averaging etc. without leaking pull logic everywhere.

Additionally use of this can make the intent of not-changing-pull explicit, where getDigitalValue() may or may not manipulate pull depending on a presently unknowable state, and the only alternative presently is to force the pull explicitly. This creates potential unexpected behaviours from reordering init sequences or maybe even race conditions in extreme cases.

finneyj, 2016-06-23:
@cefn 

Not quite sure I follow your argument here. I think the semantics are fairly clear - as per mbed conventions, pull logic only applies to digital inputs. When a pin enters that mode the pull-up/down of the pin will be configured according to the mode last requested. if no mode is requested, you get the default as set in the compile time configuration options.

if you enter a different pin mode (e.g. a digital output) on the same pin subsequently, the behaviour of the pin will be whatever the underlying hardware does.

Am I missing something?

cefn, 2016-06-23:
I think the corner case I'm imagining is actually excluded. 

What I had in mind was that someone might want to query getDigitalValue (e.g. in a logging library) while maintaining a separation of concerns, (i.e. if setting pull was intended to be managed elsewhere, then the implication that the first call to getDigitalValue() always assigns a pull could be worked around by explicitly asking no pull change to happen). 

However, I think this is implicitly worked around by either...

1) getDigitalValue()  IS called with a pull argument or setPull IS explicitly called in application code
2) getDigitalValue() is NEVER called with a pull argument, and setPull is NEVER explicitly called in application code

...and in either case, a 'naive' call to getDigitalValue would be harmless, as it would either be overridden by a later explicit call 1) or have no effect 2).

closed 2016-06-23 by cefn