lancaster-university/microbit-dal #164: buttonWasPressed, pinWasPressed

pelikhan, 2016-06-09:
It would be great to have the same API as micropython for button presses:

```
# returns True or False to indicate if the button was pressed since the device
# started or the last time this method was called.
button.was_pressed()
# returns the running total of button presses, and resets this counter to zero
button.get_presses()
```

It's particularly usefull in a tight loop when button presses are missed/.

http://microbit-micropython.readthedocs.io/en/latest/microbit_micropython_api.html?highlight=was_pressed

finneyj, 2016-06-09:
thanks @pelikhan 

Sure, that would be a trivial but useful enhancement. I hate to introduce duplication though: @dpgeorge - would upstreaming this API be a welcome move, assuming we preserve semantics with your API of course?

dpgeorge, 2016-06-09:
I thought the point of having the message bus with primitives like BUTTON_EVT_DOWN was so that we could easily implement such semantics ourselves (which is what we do, we listen for EVT_DOWN and use that to update the state about button presses).  Adding wasPressed would complicate the button API in the DAL.

finneyj, 2016-06-23:
@dpgeorge @pelikhan 

For sure, the aim is to allow folks to use the building blocks here to create the best programming experience possible for higher level languages. It also makes sense to consider sharing best practice round too though, where it make sense to do so to reduce effort and duplication. 

For most drivers we support calls for basic synchronous and asynchronous use, and for microBitButton we already have isPressed(), so I'm happy either way on this one.