lancaster-university/microbit-dal #165: radio RSSI clarification

pelikhan, 2016-06-09:
Could we clarify the units and range returned by the RSSI in the radio API? It is not clear that they are in ``-dbm``. However, if a student searches "wikipedia RSSI" on the web, it lands to wikipedia

https://en.wikipedia.org/wiki/Received_signal_strength_indication

which states that higher RSSI is better.

finneyj, 2016-06-23:
Hi @pelikhan 

We're just forwarding on the value received from the Nordic hardware here.  Checking the datasheet, it's kindof quirky:

```
RSSI sample result. The value of this register is read as a
positive value while the actual received signal strength is a
negative value. Actual received signal strength is therefore as
follows: received signal strength = -A dBm
```

This explains the inversion you're seeing. Essentially, if the module reports "75" it actually means "-75dbm"...

It's tempting to flip this in the driver before too many people are using it, so we can use SI units (as we do everywhere else when possible), but this would be a bit of a breaking change. Perhaps better addressed simply through documentation...

pelikhan, 2017-01-17:
@finneyj what is the final word on this? As we get more usage of radio, it keeps poping up.

finneyj, 2017-01-17:
@pelikhan - thanks for the nudge.

This is really a bug - we *should* be returning SI units. Let's take advantage of the beta on microbit.org to flip this. Such that -75dbm is actually reported as -75. This is both more accurate and more intuitive as a smaller number then relates to a weaker signal (e.g. -100 vs. -45).