lancaster-university/microbit-dal #168: Pairing mode to be selectable from config.json

bluetooth-mdw, 2016-06-15:
Hi, could we move the pairing mode (no, security, passkey, just works) to config.json please? I'm hoping it can become selectable from one of the new on-line editors and having it in config.json will make that work much easier.

Thanks

bluetooth-mdw, 2016-06-17:
Change of heart on this; Looking at how PXT works, placing this selector in config.json may not be a good idea. If I understand correctly, for this to be a Block I can select, we need to do this in C/C++ rather than a config setting. Can we instead have a method call that selects the pairing method (with a suitable default of course).

uBit.bleManager.setPairingJustWorks()

for example.....

Cheers

finneyj, 2016-06-23:
sounds fine. Would suggest a setPairingMode() method with a parameter to define one of those options. We could use a set of #defines for the enumeration, to be consistent with other earlier requests. Sound OK?

bluetooth-mdw, 2016-06-23:
Sounds A-OK.