lancaster-university/microbit-dal #169: Remove redundant items from MicroBitConfig.h

bluetooth-mdw, 2016-06-15:
Some of the constants in MicroBitConfig.h have been superseded by properties in config.json. Can we remove those which are now redundant please?

Ta very much.

finneyj, 2016-06-23:
Errr... No. :-)

MicroBitConfig options still provide the definitive defaults. The yotta config options *can* override these defaults, if set. However it's good the leave the defaults in MicroBitConfig.h to support other build systems (e.g. mbed.org)...

OK?

bluetooth-mdw, 2016-06-23:
You're the boss :-)

Perhaps some form of electric shock therapy will teach me these values can be set in two places. I keep being tripped up by thinking I've set a certain setting only to find it was being over-ridden by config.json all along.

finneyj, 2016-06-23:
Yes, that was annoying fo rmany (caught me out once too). Latest version of the ```microbit``` repo no longer overrides all config options. Replaced by a README.md that defines all options that *can* be used. So unless you now actively create an option in config.json, you'll get the default from MicroBitConfig.h (which I think is a lot clearer).

DavidWhaleIET, 2016-06-23:
Sometimes just a signpost in both places, is all that is needed to prevent
people tripping over this common problem repeatedly.

If not done already, a comment above the appropriate section in each file,
mentioning the other file, and where to check the full scheme (in the
readme) would help, no?

finneyj, 2016-06-23:
Yes, good suggestion @DavidWhaleIET.

Pretty sure this is already in the yotta configuration README (but I'll check), and is covered in the docs here: http://lancaster-university.github.io/microbit-docs/advanced/#compile-time-configuration-options

Having a reference in MicroBitConfig.h too would be a really good idea though, and there's nothing in there at the moment...