lancaster-university/microbit-dal #170: Magnetometer bearing over Bluetooth has gone wonky

bluetooth-mdw, 2016-06-17:
At version 2.0.0 rc2 the Magnetometer Bearing characteristic worked correctly, sending appropriate values in degrees as Bluetooth notifications as the micro:bit was rotated in the horizontal plane.

See http://bluetooth-mdw.blogspot.co.uk/p/bbc-microbit.html#compass 

At version 2.0.0 rc3 onwards however, it no longer works. Values are sent but they're "stuck" in the at around 350 degrees, fluctuating only by a few degrees even when rotating the micro:bit through a complete circle.

finneyj, 2016-06-23:
Hmmm.... no changes to BLE services or MicroBitCompass in between rc2 and rc3. 

There were some changes to MicroBitStorage though... 
https://github.com/lancaster-university/microbit-dal/commit/dc9b0078952bee63709771853c3ba73d08dc85a6

https://github.com/lancaster-university/microbit-dal/commit/3b435c0d30bd261de037670700ded513ab4248a9

Perhaps calibration data is no longer being stored accurately... Is this reproducible? Is it only wonky over BLE or also vi the API?

bluetooth-mdw, 2016-06-23:
I believe it is, yes. I can supply hex files from rc2 where it works fine and one from rc3 where it consistently does not. I haven't delved into it's behaviour independently of Bluetooth.... a good thought. I'm not likely to be able to look at that for a little while though. If I get a chance I will but it won't be this week I'm afraid.

finneyj, 2016-06-23:
ok, thanks @bluetooth-mdw. No worries - I'll take a look at this one.

bluetooth-mdw, 2016-07-10:
Also... if you flip the micro:bit on its back you get different values from the magnetometer bearing characteristic.... it's still facing the same direction wrt magnetic north so I would not have expected this. For example, the following readings were with the  display face down:

278
279
279
278
280
279
279
277
282

but flip it so the display is face up and the values immediately change to 

85
85
84
85
85
85
85
84
85
85
85
85
86
85

Weird, on?