lancaster-university/microbit-dal #172: Generalizing ROW1 and COL1 for nrf51-DK target

Darzu, 2016-06-20:
When building for the "nrf51dk-gcc" target ([nRF51-DK board](https://developer.mbed.org/platforms/Nordic-nRF51-DK/)), I get an error that COL1 and ROW1 aren't defined in the file /inc/drivers/MicroBitMatrixMaps.h.

There's a comment in the file saying:
```
//ROW1 and COL1 are defined in mbed classic:
//https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_MICROBIT/PinNames.h
```

As a work around, I defined ROW1 as p13 and COL1 as p4.

carlosperate, 2016-06-21:
Are you sure you have set your target correctly? `yt target bbc-microbit-classic-gcc`

finneyj, 2016-06-23:
Thanks guys. @carlosperate - I think Daryl is experimenting with other hardware platforms to see what breaks... I guess he found something. :-)

it is worth noting that this is the only item that is specific to the bbc-microbit* targets. Otherwise, we can compile on any nf51 mbed target. It would be good to abstract this away, as there's no *requirement* to run a MicroBitDisplay...

Suggest we stop using ROW1 and COL1, and instead define the constants needed for the micro:bit here: https://github.com/lancaster-university/microbit-dal/blob/master/inc/drivers/MicroBitMatrixMaps.h#L128

Then, mbed targets can define (if they wish) a display geometry with a YOTTA_CFG option. If not, then the parameterise constructor to MicroBitDisplay can handle other cases: 
http://lancaster-university.github.io/microbit-docs/ubit/display/#microbitdisplay-uint16_t-id-const-matrixmap-map

jamesadevine, 2016-07-15:
Corrected by 	ff54797, this will be in the next tagged release.

jamesadevine, 2016-07-15:
Thanks @Darzu ! :smile:

closed 2016-07-15 by jamesadevine