lancaster-university/microbit-dal #174: PacketBuffer size

Elliot-Wright, 2016-06-27:
Had issue with sending data via a PacketBuffer.

Documentation states I can send and receive up to 32 bytes at a time.

However the packet header takes up some of those bytes causing some of my data to be lost.

How many bytes are used by the header and how many are available for the payload?

jamesadevine, 2016-06-28:
Hi Elliot,

Have you got a code sample?

Have you verified what bytes are being sent and received? If so, what are your findings?

Best,

James.

alennie1993, 2017-06-13:
Hey, just going through all the issue... Did Elliot get reply to you James?
Cheers

jamesadevine, 2017-06-13:
@alennie1993 nope - sorry!

alennie1993, 2017-06-13:
Okay no problem, I'll add it to my list to have a play around with