lancaster-university/microbit-dal #175: Radio receive - event queue jams due to unprotected queueDepth decrement

whs-che, 2016-06-28:
MicroBitRadio.cpp

Variable queueDepth is manipulated in the radio ISR , via queueRxBuf(), and by an idle thread. This can cause the queueDepth to decrement to 255 and hence break the max buffers detection logic. 

i.e. idle thread reads current value of queueDepth  -> interrupt routine gets in and successfully increments the count -> idle thread resumes and promptly overwrites the correct value. Since there are now "excess" packets in the queue, subsequent decrements drive the counter to 255. Since this is larger than MICROBIT_RADIO_MAXIMUM_RX_BUFFERS no more queuing can take place. 
A possible method of solution is shown below and appears to work and also encompasses the list manipulation just to be safe. Without more in-depth knowledge of the DAL I can't comment on potential side-effects with this fix. Is there a better way to serialise access here?

Failure on test system (7 microbits broadcasting at 100ms) occurs within 30mins. Current test with the fix is at 36 hours and still okay.

```
FrameBuffer* MicroBitRadio::recv()
{
    FrameBuffer *p = rxQueue;

    if (p)
    {
        NVIC_DisableIRQ(RADIO_IRQn);  // Added - protection from isr

        rxQueue = rxQueue->next;
        queueDepth--;

        NVIC_EnableIRQ(RADIO_IRQn); // Added - re-enable isr

    }

    return p;
}
```

jamesadevine, 2016-06-28:
@whs-che Great catch! 

That also looks like a suitable patch :smile:

Is there any way you could form this into a pull request?

finneyj, 2016-06-29:
Hi @whs-che

Yes, a classic race condition that sneaked in. Thanks for the diligent testing! 

Agree this is a suitable fix. Technically NVIC_DisableIRQ isn't always safe (if soft device is running, there's a n alternative wrapper function), but as Radio and SoftDevice are currently mutually exclusive, we're fine here I think.

Let's wrap it up in a Pull Request and merge it.

jamesadevine, 2016-07-15:
Fixed by PR #176

closed 2016-07-15 by jamesadevine