lancaster-university/microbit-dal #177: The connection has timed out unexpectedly.

citysite102, 2016-07-01:
Hello, we've tried to develop micro:bit with iOS and we have encountered some weird issues after the  connection is built. It'll unexpectedly disconnect with micro:bit board and we can only get some error information like `Code=6 "The connection has timed out unexpectedly."`. Is there any possible solution or suggestion on the micro:bit firmware side/iOS side to avoid this error?

finneyj, 2016-07-01:
Hi @citysite102 

This could perhaps be related to some of the security settings.

Can you let us know what version of the code is running on the micro:bit? i.e. how the HEX file that the micro:bit is currently running was generated?

citysite102, 2016-07-01:
This is the firmware for our testing App which is built by YOTTA, and the version of microbit-dal is v2.0.0-rc4.
```
#include "MicroBit.h"
#include "MicroBitSamples.h"

#define HELLO_APP_LOOP_SLEEP (500)

#ifdef MICROBIT_HELLO_APP

MicroBit uBit;

int main() {
    
    // init the micorbit:
    uBit.init();
    
    // setup the ble services:
    new MicroBitAccelerometerService(*uBit.ble, uBit.accelerometer);
    new MicroBitButtonService(*uBit.ble);
    new MicroBitIOPinService(*uBit.ble, uBit.io);
    new MicroBitLEDService(*uBit.ble, uBit.display);
    new MicroBitTemperatureService(*uBit.ble, uBit.thermometer);
    
    uBit.display.scroll("Hello");
    
    while(1) {
        uBit.sleep(HELLO_APP_LOOP_SLEEP);
    }      
}

#endif
```

finneyj, 2016-07-01:
Thanks - looks fine to me.

You have a fair few services in there though - have you increased the GATT table size through config.json or MicroBitConfig.h? The defult size will likely be too small to hold this many BLE services...

There's some information here about this:
https://lancaster-university.github.io/microbit-docs/advanced/#compile-time-configuration-options

citysite102, 2016-07-01:
Really thanks for your response.
We already increase the GATT table size 0x700 to hold these BLE services, but we still encounter the problem..

citysite102, 2016-07-01:
I think 0x700 is the max value of the GATT table size, is there any other possible setting that may induce this issue?

finneyj, 2016-07-01:
0x700 is indeed the maximum - and this should be enough to hold the services you have here.

Hmm.... IOS has been pretty stable, as there's less variability in the BLE stacks than on other platforms (All IOS devices run the same stack really).

Have you tried deleting the bond on your IOS device and repairing?

jamesadevine, 2016-07-01:
@citysite102 Have you tried increasing the default range of the transmission power? 

We have it set to the minimum level for the children (stipulated by the BBC), the config option you will need is `MICROBIT_BLE_DEFAULT_TX_POWER`.

citysite102, 2016-07-01:
@finneyj  @jamesadevine Thx really 👍, the connection state works much better by setting MICROBIT_BLE_DEFAULT_TX_POWER to 1-2.

jamesadevine, 2016-07-01:
@citysite102 6 might be significantly better! :smile:

finneyj, 2016-07-01:
I like 7... MORE POWER!!!!   :-D

Really, 6 should be the default for most builds I think - that's 0dbm. 7 is +4dbm. Anything less than 6 is effectively attenuating the radio...

jamesadevine, 2016-07-15:
Closing as this issue is resolved.

closed 2016-07-15 by jamesadevine