lancaster-university/microbit-dal #180: Non-USB Serial Problems

Drmikej, 2016-07-07:
I'm finding it hard to believe that this is a bug but there is something odd going on. 
I'm trying to use the Serial object with general GPIO pins. 
The simplest program I've tried is:
  MicroBitSerial Serial = MicroBitSerial(P0_1, P0_2);
    Serial.baud(115200);
    Serial.send("Test\r\n");
    Serial.send("Test\r\n");
There are three issues.
1) There is always an incorrect first byte that shows as a frame error on the logic analyser.
2) The docs show that P0_1 should be tx but the signal appears on P0_2 i.e. P0_2 seems to be tx.
3) Only the first send seems to work - there is no sign of any other data after the first send.
I might be missing something really obvious but I've spent most of the day trying out variations - delays, using raw_serial etc (probably far too many variation) but can't make it work.

remay, 2016-07-07:
Regarding the pins you need to realise that the mbed pins (the pins on the SoC) and the micorbit edge connector pads are not the same.  I suspect you want:
`MicroBitSerial Serial = MicroBitSerial(MICROBIT_PIN_P0, MICROBIT_PIN_P1);`
See [Microbitpin.h](microbit-dal/inc/drivers/MicroBitPin.h) for the mappings between the mbed definitions (P0_X and the microbit edge connector)

Can't help with the rest though.

Drmikej, 2016-07-07:
Yes that makes perfect sense - and I should know well enough not to use the mbed pins by now. 
That at least clears up one aspect of strange behaviour - it was mine :-)

remay, 2016-07-07:
DOn't know if it's anything to do with your problem, but [the docs](https://lancaster-university.github.io/microbit-docs/ubit/serial/) don't show a signature for `serial.send(char *)`.

The closest is `serial.send(unit8_t* buffer, int bufferLen);`.

Does changing your code to
`Serial.send("Test\r\n", 6);`
make any difference?

[As an aside I'm a bit surprised that your code compiles, but I haven't investigated where such a signature mightbe supported]

finneyj, 2016-07-07:
Hi @Drmikej.

People with a Dr in their name are never strange - they're eccentric! There's a huge difference! :-)

A few quick observations that might help:

@remay is quite right - please do use MICROBIT_PIN_XX to address the edge connector, otherwise you do indeed get the processor pin mapping from mbed (which is not normally what you want!)

The function you will be calling here will be this one: https://lancaster-university.github.io/microbit-docs/ubit/serial/#int-send-managedstring-s

ManagedStrings are reference counted immutable strings we use in the runtime - they automatically upcast themselves from a char *... so every time you see ManagedString in a signature, you can just pop in a string literal, or null terminated char * buffer and c++ will work some black magic voodoo to ensure there are no memory leaks within the runtime.

I too have seen glitches on the serial line immediately after power up - this happens with the native mbed Serial classes... it's not uncommon to get the odd stray bit on the serial line of any device when it powers up, but I wonder if it might be worth us having a review of the ntf51822 Serial implementation to see if it can be cleaned up at all.

The losing of data is very odd behaviour though.  I wonder if you could try two things:

1) Can you share the whole code you're running, so we can have some context? Also knowing the version of microbit-dal you're compiling against would be very useful. There have been some changes in there recently.

2) I wonder if it isn't the second send that's lost, but the first... as the data is the same in both cases, it's impossible to tell. Can you try a set of back to back sends printing different outputs, and see which sends are noticed?

finneyj, 2016-07-07:
hmm... Just digging - this could be a duplicate of #149.

@jamesadevine fixed something like this a little while ago - and perhaps it hasn't made its way through to you build environment?

Drmikej, 2016-07-08:
Ok - thanks for all the input (I'm impressed and I hope this isn't  all  because I made a mistake yesterday).
After making some changes etc the status is:
1) The framing error is every time - not a power on glitch which I would happily ignore.
2) no problem with pins - the bug was me.
3) Although there is no send with the right signature string is usually converted to managed string. I tried Serial.send("Test\r\n", 6); and it doesn't compile. It is the second send that is failing (confirmed by sending different strings and commenting out one at a time)

The full program was:
#include "MicroBit.h"
MicroBit uBit;
int main() {
  uBit.init();
    MicroBitSerial Serial = MicroBitSerial(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
    Serial.baud(115200);
    Serial.send("Test\r\n");
    Serial.send("Test\r\n");
    uBit.sleep(2500);
}

When I take away the uBit.init() it works. I get both sends on the logic analyser. 
Presumably if I create a MicroBit object I have to use its Serial member.
 There doesn't seem to be any documentation of redirect?
So I attempt to use uBit.serial:
#include "MicroBit.h"
MicroBit uBit;

int main() {
    uBit.init();
    uBit.serial.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
    uBit.serial.baud(115200);
//     printf("Hello World");
    uBit.serial.send("Test\r\n");
    uBit.serial.send("Test\r\n");
    uBit.sleep(2500);
}

Nothing shows on the logic analyser except the usual frame error byte. If I put the printf in then Hello World appears on pin 1 nothing else. If I take out the redirect then I see what I expect on the USB serial.

As to the Dr being "eccentric". Sorry folks my name was taken in GitHub and I made a mistake in picking an alternative. Alerting people to you eccentricity is never a good idea :-) 
If I ever find a way to change it I will.
mikej

jamesadevine, 2016-07-08:
What version number are you on in the DALs module.json?

James.

Drmikej, 2016-07-08:
This is what you always ask me :-)

  "dependencies": {
    "microbit": "^2.0.0-rc3"

jamesadevine, 2016-07-08:
The most important information! :)

This was fixed as of 179b32f, you will need to update to rc4!

Best,

James.

Drmikej, 2016-07-08:
I upgraded to rc4 as soon as you asked about what version I was using.
No change in behaviour.
The printf is visible on the logic analyser but the sends don't do anything with or without the printf.
#include "MicroBit.h"
MicroBit uBit;

int main() {
    uBit.init();

    uBit.serial.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
    uBit.serial.baud(115200);

    printf("Hello WOrld");
    uBit.serial.send("Test\r\n",SYNC_SPINWAIT );

    uBit.serial.send("Test\r\n",SYNC_SPINWAIT );

    uBit.sleep(2500);

}

finneyj, 2016-07-08:
hmmm.... I just ran this against a vanilla microbit-dal-rc4:

```
MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.serial.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2);
    uBit.serial.baud(115200);

    // Insert your code here!
    while(1)
    {
        for (int i=0; i<10; i++)
        {
            uBit.display.print(i);

            uBit.serial.send("Test\r\n");

            uBit.sleep(2500);
        }
    }

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}
```

I get visible data that looks plausible squirting out of P1...

finneyj, 2016-07-08:
p.s. I don' t have a logic analyser handy, but can see a square wave at the right frequency on my oscilloscope.

Drmikej, 2016-07-08:
I copied and pasted your program. 
It just hangs and I get one pulse - the byte with an error frame.
If I take out the send then I see the loop on the LED display.

I am now taking everything to pieces and starting over. 
There is something odd with my setup.
And I can hardly remember how to use a scope :-)

Drmikej, 2016-07-08:
Taken everything off the micro:bit that has nothing to do with the "experiment" - it is now a naked micro:bit with just pins 1 and 2 connected to a proto board and a logic analyser. .
I checked the version in the json file and its 
 "dependencies": {
    "microbit": "^2.0.0-rc4"
  }
}
Is there any way I could be compiling against something else by accident? I did a yotta build to make sure. (I usually use NetBeans to do the build). I've also done a few clean builds etc.. 
Was going to get scope on the pin but as nothing displays on the LED I think we can assume that its hanging at the send and hence no pulses. to see, 
I've even disconnected the logic analyser and I still don't see the display loop on the LED.
It has to be a software difference.

finneyj, 2016-07-08:
yes, it does sound like something in your toolchain. Just to confirm - here's the binary I used. This should give you a positive test result if your hardware is ok.

Looking a your module.json, you have a dependency on the correct version of microbit, but I wonder if this perhaps didn't cascade down to the microbit-dal module, which is the one you need to be on rc4?

[microbit-samples-combined.zip](https://github.com/lancaster-university/microbit-dal/files/354233/microbit-samples-combined.zip)

Drmikej, 2016-07-08:
Big thanks for the zip - it worked!
And proved that I must have problems with an out of date microbit-dal.
So I:
yotta uninstall microbit
then I deleted all of the related folders in the project (I have not had to do this before)
then 
yotta install lancaster-university/microbit
Followed by a yotta build
then (just in case) a NetBeans clean build
and it works.

Thanks everyone - sorry it wasn't a real bug but also glad it wasn't a real bug (if you see what I mean).
Must remember to delete folders when upgrading.
Only got the strange frame error on the first byte to worry about now - and thats really not much of a problem.

finneyj, 2016-07-08:
great stuff - thanks for reporting back. Closing this one off.

p.s. When I said "eccentric" I meant it as a compliment - you should totally keep your github handle! :-)

closed 2016-07-08 by finneyj

Drmikej, 2016-07-08:
I took it as one but it doesn't mean that it is a good idea to let everyone know that you have an advantage :-)
The thing that amazes me is that I have never met or encountered online another programmer with the same name as me but my name is nearly always gone when I try to sign up to anything ! :-(