lancaster-university/microbit-dal #182: Enhancement request - support PWM via bluetooth IOPinService

cpseager, 2016-07-14:
Objective - to use the microbit as a bluetooth games controller for student Scratch programs.

[image]

Hardware - microbit fitted with the Kitronik MI:power board, as it provides a neat 'easy to hold' unit.

Firmware - life is too short for pairing so I am currently using 'microbit-2-0-0-rc2_pwr6_open.hex' from @bluetooth-mdw personal blog page (thanks!).

Already working with Scratch - (see screenshot) bearing, tilt (accelerometer), temperature, buttons, msg display

Not working - producing different sounds on output 0 (piezo is included on the Kitronik power board)

Using the existing IOPinService directly AFAIK you can only produce a fixed 'beep' when pin0 is configured as an analog output. There is no way to change the analog period via this service? I currently make the beep like this:

```
var pin:number = 0;
var value:number = 512 / 4; //value is *4 within microbit
return this.writeCharacteristicValue(MICROBIT_IO_PIN_DATA, [pin, value])
.then(()=>{
  return makeDelay(100)
  })
.then(()=>{
  return this.writeCharacteristicValue(MICROBIT_IO_PIN_DATA, [pin, 0x00]);
});
```

Although this works it does not allow 'music' or even various different 'beeps'.

I do realise I could probably write my own custom event using the bluetooth EventService, but it seems far more logical to be able to support pwm (for sound/motors/servos etc) directly via the IOPinService? This would seem fairly trivial to do - just an extra 'MICROBIT_IO_PIN_PWM' characteristic, which could then be used like this for the sound frequency 'freq':

```
var pin:number = 0;
var value:number = 512;
var valueMsb:number = Math.floor(value / 256);
var valueLsb:number = value % 256;
var period:number = Math.floor(1000000 / freq);
var periodMsb:number = Math.floor(period / 256);
var periodLsb:number = period % 256;
var durMsb:number  = Math.floor(duration / 256);
var durLsb:number  = duration % 256;

return this.writeCharacteristic(MICROBIT_IO_PIN_PWM, [pin, periodLsb, periodMsb, valueLsb, valueMsb, durLsb, durMsb]);
```

I haven't studied the DAL in great detail but at a glance I think something like this snippet could work for the new characteristic?
	
```
 struct IOPwmData
 {
   uint8_t     pin;
   uint16_t    period;
   uint16_t    value;
   uint16_t    duration;
 };
		
 io.pin[pwmData->pin].setAnalogPeriodUs(pwmData->period);
 io.pin[pwmData->pin].setAnalogValue(pwmData->value);
 if (pwmData->duration > 0) {
   fiber_sleep(pwmData->duration);
   io.pin[pwmData->pin].setAnalogValue(0);
 }
```

Any reason why this wouldn't work or why pwm is not currently supported already? Thanks

jamesadevine, 2016-07-14:
Hi @cpseager,

That is one very cool objective!

It does appear that PWM functionality has not been implemented, and we should open up a discussion with @bluetooth-mdw just in case there was a reason behind omitting it.

Your code snippet looks like a step in the right direction. Is there any reason why you would need the `duration` field? Could you just turn it on/off from the program?

cpseager, 2016-07-14:
Hi - for the duration I was thinking about wrapping the snippet within a

`while (len >= sizeof(IOPwmData))`

type loop (as the data characteristic already does) so that you could send multiple sounds (e.g. 'ding-dong') in one write. But it's not essential, my fixed tone example above uses promises to stop the tone and that works fine too.

bluetooth-mdw, 2016-07-25:
Just back from holiday.

Oh my. @cpseager ..... support in Scratch looks bloomin' wonderful! :-)

There's no reason we didn't provide direct BLE support for PWM... it just wasn't something we thought of. Looks like a very good idea though. 

I'd like to draw LU's attention to https://github.com/lancaster-university/microbit-dal/issues/185 in the context of this issue though... thought I'd submitted this ages ago but found today I'd only filed it in my head. Anyway.... this also looks like a candidate for being an optional characteristic.

If we do go ahead with this addition to the IO Pin Service... which I hope we will.... let me know and I'll update the documentation, liaising with LU and @cpseager. 

Good to come back to this!

bluetooth-mdw, 2016-07-26:
@cpseager Hi, could you connect with me via Twitter? I'd like the chance to discuss your work in this area further. Looks great! I'm @bluetooth_mdw on Twitter. Thanks.

cpseager, 2016-07-26:
@bluetooth-mdw FYI cpseager on twitter is not me! Have just emailed you on your gmail account used to publish the android app.

bluetooth-mdw, 2016-07-27:
@cpseager Oh! LOL there's a very confused @cpseager on Twitter with a weird message from me then :-)

Thanks!

bluetooth-mdw, 2016-08-25:
I'd like to get some opinions on the following with respect to use cases / typical requirements.

We have three pins that support PWM. In my mind it would be nice to be able to address all three in one write operation, giving a form of polyphony for music applications. micro:bit playing chords. There could be whole micro:bit orchestras!

In consultation with @finneyj it's been agreed that the client will be responsible for sending "OFF" messages so that duration is not something the profile need concern itself with. This saves memory and the need to consume fibers to perform concurrent execution of on|wait|off sequences on pins concurrently.

That leaves a characteristic which I would like to contain an array of 1, 2 or 3 instances of this struct:

struct IOPWMData
 {
   uint8_t     pin;
   uint32_t    period;
   uint16_t    value;
};

That's up to 3 x 7 octets = 21 octets.

But unfortunately this will not fit. The maximum size of payload the nRF51 can handle is 23 octets but this includes the ATTribute protocol opcode and length, which total 3 octets. So we're one octet short!

I'm assuming both period and/or value might vary each time the client writes to the pin or at least we need that flexibility.

Question: the types shown come from the DAL APIs. Do we really need full microsecond granularity for the period? Or can I shave a byte off that field so that we come in with 3 x 6 + 3 = 21 octets payload max size for this?

Thanks guys

cpseager, 2016-08-25:
For maximum flexibility I suggest thinking about different PWM use case (servos, motors etc) not just sound.

For music you want a fixed duty cycle and change the period. For servos you want a fixed period and to change duty cycle. For a DC motor you may want to 'trim' its speed by tweaking either.

The 'x4' in the existing characteristic already restricts resolution considerably and struck me as being a bit odd, so I think the new characteristic should be able to support full resolution.

Also it's only a quirk at present there are 3 PWMs - I think if #115 is resolved there could actually be 4?

So given the restricted payload size my vote would be for full period/duty cycle resolution on 2 instances rather than partial resolution on 3! Thanks Martin!

martinwork, 2016-08-25:
I think I read that the period is shared by all PWM channels. I hope I have misunderstood! But that would mean you don't need a period number for each channel.

finneyj, 2016-08-25:
the mbed nrf51822 PWM implementation does indeed have a shared period for all PWMs... A limitation of the way PWM is implemented I'm afraid. Suggest we don't constrain APIs to this though, just in case future versions of the microbit have a cpu with better support for PWM...

bluetooth-mdw, 2016-08-26:
Thanks everyone. @martinwork I didn't know that! You learn something new everyday etc :-)

I do have an alternative but I'm reluctant to go there. "Long reads" are defined and involve making multiple reads, each one specifying an offset into the value you're reading. I'm reluctant to use this because support for long reads is optional in clients.

As things stand I'm very much inclined to go with @cpseager 's suggestion I only support a maximum of 2 PWM pins at a time.

Speak now or forever hold your piece!

finneyj, 2016-08-26:
sounds ok to me - you could issue back to back send operations if you need to write to >2 pins I guess (and most likely implementations will do that for each pin anyway, as it's simpler!).  :-)

bluetooth-mdw, 2016-08-26:
I'll carry on with this next week.... should give people time.

@cpseager on use cases... .yes, absolutely. Was just using sound as an example.

A good test will be whether or not I can make m:b play a convincing Am7 from a smartphone app!

cpseager, 2016-09-02:
Added in #209 

Thanks @bluetooth-mdw

closed 2016-09-02 by cpseager

finneyj, 2016-09-02:
Indeed - nice job @bluetooth-mdw.

bluetooth-mdw, 2016-09-02:
Gracias amigos :-)

Don't forget the associated documentation updates: https://github.com/lancaster-university/microbit-docs/pull/33 

Cheers