lancaster-university/microbit-dal #184: netbeans reports yotta_cfg_mappings.h as missing include in MicroBitConfig.h

drmpf, 2016-07-18:
Compiling the BLE uart example, netbeans reports
 yotta_cfg_mappings.h as missing include in MicroBitConfig.h
changing 
#include "yotta_cfg_mappings.h"
to 
#include "../platform/yotta_cfg_mappings.h"
resolves the issue, but not sure why it was not found.

jamesadevine, 2016-07-18:
@drmpf That's weird! And it has no troubles with other headers?

drmpf, 2016-07-18:
Yes, I agree, weird.  No trouble with other headers, except the [missing sleep_api.h header](https://github.com/lancaster-university/mbed-classic/issues/1)

drmpf, 2016-07-18:
As noted in  [missing sleep_api.h header](https://github.com/lancaster-university/mbed-classic/issues/1) this is a Netbeans code include processing issue.  It does not effect building only code navigation within Netbeans. 
So closing this issue also.

closed 2016-07-18 by drmpf