lancaster-university/microbit-dal #187: Pairing process recommendations for iOS and OS X

sandeepmistry, 2016-07-27:
Is there a recommended pairing procedure for iOS and OS X?

Unlike Android, when the micro:bit is in pairing mode it does not show up in the iOS and OS X Bluetooth settings. Only BLE peripherals that advertise or solicit certain UUID's like HID and ANCS will appear.

One workaround is to put the micro:bit pairing mode, then use an generic BLE app like LightBlue to discover the device and try to read a characteristic. This is not a great out of box experience.

Is there a better way or something I am missing?

smartyw, 2016-07-27:
There isn't a better way with iOS that I know of and the use of a "pairing mode" in micro:bit is by design. We wanted / needed to ensure micro:bit use could not be blocked in classroom scenarios by the simple act of someone else connecting to your micro:bit. So once paired, white list filtering ignores connection requests from peers which are not paired and the only time the device advertises and allows connections from misc devices is when deliberately put in pairing mode. 

The 'better way' is to use Just Works pairing which simplifies the procedure but still involves pairing mode or no pairing at all if security requirements allow this. It's really important to remember that micro:bit was created in the first instance for children to use, often in group scenarios.

martinwork, 2016-07-27:
For iOS, you can use the BBC micro:bit app to do the pairing with the Connections option.

remay, 2016-07-27:
I've always found it a bit odd with the current pairing requirements that the central app needs to attempt reading a characteristic in order to initiate the pairing process.  On connection from a non-paired central (in pairing mode) the micro:bit peripheral should, IMHO, issue an SMP pairing request, forcing the pairing to occur immediately.  Whilst not solving any on the iOS/OSX issues mentioned here, it would make the process slightly easier (and although I am not familiar with noble, I do believe that it would then be relatively easy for a noble-based app to monitor for the SMP pairing request and perform the pairing).

bluetooth-mdw, 2016-07-27:
That would make it a teeny bit easier but you'd still need *something* to make that connection.

To be honest the way to go and what I've been discussing with @finneyj for quite some time is to support  no pairing/just works/passkey as the options in such a way that this can be exposed to tools like Blocks and PXT and then for those scenarios where security is a barrier rather than benefit you can use less or no security. 

I have a PR in for Just Works which has been awaiting processing for ages... @finneyj @jamesadevine what's the plan here? See https://github.com/lancaster-university/microbit-dal/pull/154 

Btw @smartyw is me... personal account!

jamesadevine, 2016-07-27:
@bluetooth

jamesadevine, 2016-07-27:
@bluetooth-mdw Summer is a tricky time :smile: @finn

jamesadevine, 2016-07-27:
@bluetooth-mdw Summer is a tricky time :smile: @finneyj is on holiday at the moment, and it looked like he was managing that PR (whilst i was on holiday), so i'm curious as to why it wasn't merged, perhaps due to time constraints.

I believe @finneyj was actually working on a few of these outstanding issues, so i'm hesitant to push the big green button on the Just Works pairing. 

I'm happy to merge the display interrupt code @bluetooth-mdw .

sandeepmistry, 2016-07-28:
@martinwork @bluetooth-mdw thanks for letting me know about the iOS app, I think I can use a similar approach on OS X.

Longer term, maybe the micro:bit could advertise one of the special UUID's that iOS/OS X shows in the settings screen? I think "Alert Notification Service" 0x1811 or "Immediate Alert" 0x1802 would work nicely.

bluetooth-mdw, 2016-07-28:
I'd only be happy with us doing that if micro:bit actually implemented one of those services and then only if the service was mandatory. Given memory constraints (and other reasons) I can't see us adding either of these services as mandatory. 

My personal opinion is that improving the pairing experience is best achieved though making it easier to opt to use Just Works or no pairing at all instead of Passkey based pairing. @finneyj has agreed to do some DAL refactoring of this area so that the three pairing options can be selected in a such a way that we can create PXT Blocks for pairing modes easily and then it should be a doddle for anyone to create the pairing behaviour they want.

Beyond that it is what it is. Any Bluetooth LE device which requires a secure link will behave this way when used with an Apple peer device. It's the way Apple approached this requirement of the Bluetooth specification. micro:bit is for education primarily and this is the way the world of Apple + Bluetooth is so a bit of reality won't hurt.