lancaster-university/microbit-dal #189: I2C Polling With NAK

Drmikej, 2016-07-30:
I'm working on connecting an I2C device to the micro:bit. 
It has two modes - clock stretching or polling.
I have got it working in clock stretching mode with no problems - it just worked.
For curiosity sake (sorry) I decided to try the polling mode.
In poling mode the device returns NAK when the master puts an address frame on the bus. 
Then the master is supposed to poll trying to read and being stopped by the NAK until the device is ready and it sends an ACK followed by the data. 
Tried it and it doesn't work. The first read gets an NAK but then something strange happens - the clock goes low for a long time and then floats high and not much happens after that. 
I've read some of the code and come to the conclusion that it the slave sends a NAK then the I2C is reset.
Is this correct?
If so there seems to be no way to poll for data ready.

jamesadevine, 2016-08-01:
@drmikej Hahaha, thanks for the version number :smile:

To clarify, the micro:bit is the master, and your external device is the slave?

Would you mind if i saw some code?

There are two reads, if you look at the class definition here:

https://github.com/lancaster-university/mbed-classic/blob/master/api/I2C.h

There could be subtle differences between the two, could you try the other version of read:

`int read(int ack);`

Drmikej, 2016-08-01:
The device is an HTU221D and I've used it before with other microcontrollers - Edison, Pi and Arduino with no problems. 
Yes microbit is master and as I say it all works perfectly with clock stretching and if I wait long enough for the data to be ready with polling - it seems to dislike being sent a nack.
Program is
    char buf[4] = {0xF3};
    uBit.i2c.write(0x80, buf, 1);
    while (uBit.i2c.read(0x80, buf, 3) == MICROBIT_I2C_ERROR) {
    };

the F3 is the command to send data not using clock stretching. The device simply sends a NACK every time the micro:bit trys to read some data and hence the polling loop.
What seems to happen is that the clock goes low and stays low for a while - then it goes back to as if nothing had happened. I get no data and I think the polling loop exits at once. 

As to using the other read function - I've read it though and it seems to do a reset as soon as there is an error as well but I will try it as soon as I put the circuit back together.

Drmikej, 2016-08-02:
After spending far too long on a problem that is really only a matter of curiosity as far as I'm concerned I have to admit defeat. 
I have tried int read(int ack); and its a low level function that uses further mbed functions. 
I have used ((I2C)uBit.i2c).write(0x81)  and I do get the slave to send a NACK back but when I trace though write I find:

int I2C::write(int data) {
    return i2c_byte_write(&_i2c, data);
}

So moving on to  i2c_byte_write I find:

int i2c_byte_write(i2c_t *obj, int data)
{
    int status = 0;
    if (!obj->address_set) {
        obj->address_set  = 1;
        obj->i2c->ADDRESS = (data >> 1);

        if (data & 1) {
            obj->i2c->EVENTS_RXDREADY = 0;
            obj->i2c->SHORTS          = 1;
            obj->i2c->TASKS_STARTRX   = 1;
        } else {
            obj->i2c->SHORTS        = 0;
            obj->i2c->TASKS_STARTTX = 1;
        }
    } else {
        status = i2c_do_write(obj, data);
        if (status) {
            i2c_reset(obj);
        }
    }
    return (1 - status);
}

notice the reset function if the status is set.
Moving on to  i2c_do_write 

int i2c_do_write(i2c_t *obj, int value)
{
    int timeOut = 100000;
    obj->i2c->TXD = value;
    while (!obj->i2c->EVENTS_TXDSENT) {
        timeOut--;
        if (timeOut<0) {
            return 1;
        }
    }
    obj->i2c->EVENTS_TXDSENT = 0;
    return 0;
}

It only seems to return a non zero status if the transmit times out. No distinction between timeout and NACK as in the classic mbed function.

To test my idea that the functions do not return a simple NACK response I made retries in MICROBIT_I2C.h public and set it so that the standard retries function didn't attempt to reset the I2C bus on any error before each loop. 

       char buf[4] = {0xF3};
    uBit.i2c.write(0x80, buf, 1);
     
      do{
           uBit.i2c.retries =MICROBIT_I2C_MAX_RETRIES;
      }while( uBit.i2c.read(0x80, buf, 3)==MICROBIT_I2C_ERROR);

This works in that after two reads we do get the data. First read returns NACK after the address is presented,, then there is an 83ms period when the clock is held low - no idea why - and then the second read returns ACK and the data is read. 
At this point I think i give up and accept that I2C polling is too difficult.

AndyEngland521, 2019-06-06:
@Drmikej this is an excellent write-up on a problem that I am also experiencing. Pinging this to see if polling will ever be possible. Possibly add attention to the issue. I see a 70 ms clock low after a NAK from my slave device when polling with the microbit. I also see a 5 uS bump high after every write() finishes. The next I2C transaction after this bump inevitably fails.

martinwork, 2019-06-07:
Excuse me if this is a silly question, but could this be about pullups? There seem to be questions about pullups with the HTU21D (not 221).

AndyEngland521, 2019-06-10:
@martinwork Most definitely not. Pull ups may change rise and fall times a bit, but 70 ms is far too long to be caused by the i2c pull ups. I suspected pull ups might be the issue behind my 5uS bump, however, the bump has a nice clean, square edge, which again, tells me I don't have an issue with pullups, and is probably somewhere in timing.

Drmikej, 2019-06-11:
I have to admit that this problem is now almost forgotten in the dim and distant passed but... I'd still like to know what is going on :-)
and I agree pullups aren't the problem but I can't make any sensible suggetsions.