lancaster-university/microbit-dal #191: Compiling a project with a space in the path fails

adamboardman, 2016-08-03:
Ninja fails to find NRF51822.ld as path is not escaped/quoted in parameter list

[Ubuntu+yotta]
To reproduce just move microbit-samples to a sub folder with a space in its name, eg Path Name

:~/Path Name/microbit-samples$ yt build

Gives:
arm-none-eabi-g++: error: Name/microbit-samples/yotta_targets/bbc-microbit-classic-gcc/CMake/../ld/NRF51822.ld: No such file or directory
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed

Compared to:

:~/microbit/microbit-samples$ yt build

No errors reported

jamesadevine, 2016-08-04:
@adamboardman I think this is a relatively easy patch, but i'm out and about at the moment, and unlikely to be at a machine capable of building through yotta until this evening.

If you browse to the `yotta_targets/bbc-microbit-classic-gcc/CMake/toolchain.cmake` file, and edit line 22 to be:

```
set(CMAKE_EXE_LINKER_FLAGS_INIT    "${CMAKE_EXE_LINKER_FLAGS_INIT} -mcpu=cortex-m0 -mthumb -T\"${CMAKE_CURRENT_LIST_DIR}/../ld/NRF51822.ld\"")
```

It should theoretically pass the quotes directly to the command line, where g++ should treat it as an unescaped directory string.

jamesadevine, 2016-08-05:
https://github.com/lancaster-university/microbit-targets/commit/09cf98ea0c05a5be49435cc6ad15727caeeaee62

I've tested locally and confirmed this works.

Additionally for those interested in avoiding source forge, I made an experimental target ages ago that uses a python script to merge hex files together instead:

https://github.com/lancaster-university/microbit-targets/commit/817dee7cebc6f38dadab45d8077591f91d0f2b7c

closed 2016-08-05 by jamesadevine