lancaster-university/microbit-dal #192: Add BLE characteristic(s) for light sensor

remay, 2016-08-08:
I know that the light sensor came late in the day, but shouldn't we add a BLE characteristic for it?

bluetooth-mdw, 2016-08-08:
We should, yes :-)

microbit-mark, 2017-11-23:
I have a recent request for this in support as a tool to aid teaching IoT, so adding a +1

bluetooth-mdw, 2017-11-23:
I believe the IOPinService or EventService could / should be used for this. See https://youtu.be/dyFyq42Snkw for an example of light dependent resistors accessed over Bluetooth, in this case using the Event Service as it gives you full 10 bit resolution.

jaustin, 2018-03-23:
@bluetooth-mdw digging up an old issue as I came here to file this because people are doing some music related stuff with micro:bit+Bluetooth and the light sensor might be interesting there too (https://github.com/Louismac/CBMicroBit). 

Can you elaborate on the design you imagine for this? I'd thought it would just be a new service similar to the way you did the temperature one? Ideally it'd be nice for there to be enough hints in this ticket for someone to step up and implement something.

bluetooth-mdw, 2018-04-05:
Can you clarify the idea/requirement, please? Is this about externally connected light sensors or about using the LED matrix per this info: https://support.microbit.org/support/solutions/articles/19000024023-how-does-the-light-sensing-feature-on-the-micro-bit-work-
Cheers

Martin

jaustin, 2018-04-05:
I was thinking about the 'internal' light sensor - IE the display

bluetooth-mdw, 2018-04-05:
OK, thanks @jaustin 

I looked at the DAL API and see that display.getBrightness() returns a value from 0-255. A simple approach would be to add a characteristic to the LED Service called BrightnessLevel with support for notifications (only). Subscribing to notifications should cause an automatic switch to the DISPLAY_MODE_BLACK_AND_WHITE_LIGHT_SENSE display mode and start sampling the light level at a fixed rate, sending a notification for each one.

If wanted to get all fancy, we could have  support for one off reads as well and could have a characteristic for setting the sampling period but... do we really need these features? I doubt it. KISS I think applies.

Comments?