lancaster-university/microbit-dal #196: Magnetometer calibration issues

bluetooth-mdw, 2016-08-17:
Possibly this should be two different git issues but.....

1. After apparently successfully calibrating the magnetometer by drawing a circle and seeing the smiley face displayed, then connecting to my m:b, pairing, resetting, I was asked to calibrate again. Did te act of pairing wipe the calibration data?

2. Sometimes, with the same hex file installed, calibration works, more often, it does not and a pattern appears on the display, presumably indicating 'out of memory'. I can't recreate this on demand but it's the usual behaviour I see. There must be something non-deterministic going on in there.

Both reports relate to rc4.

Hex file attached.

[microbit-2-0-0-rc4-ABDILMT-P-pwr7.zip](https://github.com/lancaster-university/microbit-dal/files/422386/microbit-2-0-0-rc4-ABDILMT-P-pwr7.zip)

alennie1993, 2017-06-13:
#78

alennie1993, 2017-06-13:
#76