lancaster-university/microbit-dal #199: Two consecutive calls to uBit.serial.send("Display Busy\n",SYNC_SLEEP) locks up micro:bit

drmpf, 2016-08-18:
This short test program locks up the micro:bit

```
#include "MicroBit.h"

MicroBit uBit;
Ticker timer;

void displayHello() {
    int rtn = uBit.display.scrollAsync("HELLO WORLD!");
    if (rtn == MICROBIT_BUSY) {
        //SYNC_SLEEP locks up if two calls made
        uBit.serial.send("Display Busy\n",SYNC_SLEEP); // ASYNC works but loose chars with more calls
        uBit.serial.send("Display Busy1\n",SYNC_SLEEP); 
    }
}

int main() {
    uBit.init();
    displayHello(); // call the first display immediately, if you leave this line out first display is delayed until timer below fires.
    timer.attach(&displayHello, 1.0); // and every second there after
    release_fiber(); // let the timer run.
}

```

My reading of the docs on SYNC_SLEEP is that it should block the fiber until the data is sent.  This does not seem to be happening.  If I replace SYNC_SLEEP with ASYNC, the micro:bit does not lock up BUT data is lost as more calls are added.

jamesadevine, 2016-08-18:
Hi @drmpf,

I will surface for this one, because it looks pretty juicy.

The mbed Timer class calls your function in interrupt context, which means there are no threads to configure for "synchronous" behaviour, which SYNC_SLEEP relies upon!

There are a couple of alternatives for the approach you are taking:

* Spawn a fiber that waits, and fires an event every 1000 ms. Have a listener for your event, which calls displayHello.

* We get some free time to implement a call that calls your function in n seconds. :wink:

For now, the first solution will have to suffice at least until I've got my current workload out of the way, and/or at @finneyj is back from holiday. (Dr.s, eh? 😛 )

drmpf, 2016-08-19:
Thanks for the quick response.  
I got suckered by uBit.display.scrollAsync( ) working.
Will try your first approach. 
When you come to look at it, perhaps implement the first approach wrapped up in a single method.
e.g.  
```
uBit.repeat(&fn, repeatInterval, initDelay);  // creates a fiber that calls fn repeatedly
//initDelay defaults to 0.  
// fn is executed on created fiber.  (no events created)
```