lancaster-university/microbit-dal #203: MicroBitSerial::eventAfter(int len) only fires once.

drmpf, 2016-08-27:
`MicroBitSerial::eventOn(ManagedString delimeters) `fires repeatedly whereas
`MicroBitSerial::eventAfter(int len)` only fires once.

Was trying to write a program to process Serial chars as they arrived.
set `uBit.serial.eventAfter(1);`  but only got notified for the first char.

If this is by design, what do I need to code to read the serial chars as they arrive, via events?
Resetting  `uBit.serial.eventAfter(1);`  inside the method is not reliable as the next char may have already arrived before the event was processed and the afterEvent() reset.  
In this case the last transmitted char would be missed.

jamesadevine, 2016-08-27:
Hey @drmpf,

I'm really glad you're putting the serial interface through its paces, it's entirely different using it rather than developing it; and it's great to have feedback from the frontline :smile:

As a result of #181 this discrepancy between behaviours will be patched. I think @finneyj and I are planning to bash through quite a few of these issues in the coming weeks.

We have a little bit of work currently underway for micro:bit that is consuming the majority of our time, so we are a bit frazzled (😅 ) but it is very important work nevertheless. :smile:

drmpf, 2016-08-27:
Look forward to it. I noticed the lazy initialization in passing, but had not fallen in that hole yet :-;

drmpf, 2016-08-28:
You might consider an additional event that fires when new data is received, as shown below.
That is an event similar to the `MicroBitRadioDatagram::packetReceived()`  `MICROBIT_RADIO_EVT_DATAGRAM` event.

```
void MicroBitSerial::dataReceived() {
    if(!(status & MICROBIT_SERIAL_RX_BUFF_INIT))
        return;
    bool sendEvent = false;
    if (rxBuffTail == rxBuffHead) 
        sendEvent = true;  // buffer was empty and received new data

    //get the received character
    char c = getc();

 .  .   .
    else
        //otherwise, our buffer is full, send an event to the user...
        MicroBitEvent(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_RX_FULL);

    if (sendEvent)  // only send event once for first new data when RX buffer empty
       MicroBitEvent(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_RX_DATA_AVAILABLE);
}

```

The check of RX buffer empty prevents event spamming if the processing method is slow to handle the event.  Also suggest you add this anti-spamming check to the MicroBitRadioDatagram also, i.e. only send event when `(rxQueue == NULL) ` 

Adding this event significantly simplifies the coding for handling incoming Serial data to

```
void handleRxData(MicroBitEvent) {
   while (uart->isReadable()) { // loop to handle all the available new data since last event processed
        char c = uart->read(); // won't block due to isReadable above
     . . .
   }
}

int main() {
    uBit.init();
    uBit.messageBus.listen(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_RX_DATA_AVAILABLE, handleRxData);
    release_fiber();
}
```
If you do add this extra event, please also add it to the MicroBitUARTService.