lancaster-university/microbit-dal #205: Magnetic field value with calibration

pelikhan, 2016-08-29:
Is it possible to get a reliable reading on the magnetometer without calibration? If not, can we document why?

gingemonster, 2016-10-21:
I presume from the lack of check-ins attached it was decided this wasn't possible or sensible? If so perhaps explaining the use case would help come up with a better solution. Sensing the microbit is on close proximity with a magnet is very usefull, for example calculating the RPM of a wheel spinning past the microbit. For these kinds of uses having to calibrate every time is really annoying as the accuracy is not important in these use cases

finneyj, 2016-10-21:
@gingemonster @pelikhan 

Totally understand, and agree about how annoying the need to calibrate can be. Whilst we could enable this functionality, there would indeed be a significant error in the values read. For your use case here (where it is the *relative* field strength that matters, you're quite right that this is less of an issue. However, at the same time, someone who does want absolute, accurate readings would be mislead.

This was parked because an upcoming release of a new firmware image for the micro:bit will allow for persistent storage of sensor calibration data (and bluetooth bonding data, accessibility data etc). So calibration would then just be a one off event (or something user driven). This seemed to meet the needs of all users...

thoughts?

gingemonster, 2016-10-21:
Hi,

The new persistent storage seems a good compromise and I think should solve the issue really, thanks for giving the details

lemio, 2020-09-17:
Hey, would it be possible to make this function available? https://lancaster-university.github.io/microbit-docs/ubit/compass/#setcalibration We're using the micro bit as some kind of Theremin, with magnets and use magnets as some way to do RFID The values don't have to be accurate, but it would be great if we can avoid the calibration. 

Thanks in advance ;)