lancaster-university/microbit-dal #206: metadata for partial flashing

mmoskal, 2016-08-30:
Not sure if this is the right repo, but I guess most issues are here.

The way we compile programs in TD, PXT, and possibly even Python, is that there is a template .hex file patched up with a much smaller user program. Provided the outer hex file doesn't change, we should be able to update just the user program. This should cut the flashing time (especially over bluetooth) dramatically.

Thus, I propose the following structure:

```
hex hash : 8 bytes
user code start: 4 bytes
user code size: 4 bytes
```

`hex hash` would be up to the application (we use first bytes of SHA of the C++ sources and a few version numbers). The flashing software could just flash user code, if the existing hex hash matches. We could probably also go for a 4 byte hash, and 2 bytes for start and size (assuming both are 16-aligned for example), but I guess this is needless complication.

I believe this metadata needs to site at a fixed address, otherwise partial flash of the device may not delete the old metadata, and the flashing software may go for the partial flash when a full one is needed. One address that would work on a Nordic device is 0x1000-16 (there is over a hundred free bytes at the end of the MBR, at least now).

Otherwise, we could use something else, defined in the C++ source code, using this trick https://mcuoneclipse.com/2012/11/01/defining-variables-at-absolute-addresses-with-gcc/

cc @jamesadevine @finneyj @jaustin

remay, 2016-08-30:
I *think* that the bootloader already supports this, and that the BLE OTA flash can also already support it - although I can't find the links that I was reading that make me think this.    For serial/usb flashing I think the usb support chip would need additional work, but again I do believe that the bootloader already has support using the SWI link ...

https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00021.html
https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00026.html

finneyj, 2016-08-30:
Yes - I've had this in the back of my mind also. Would be great to investigate this.

I'm pretty sure the USB loader does NOT support this. Standard mbed behaviour is a block erase of the device before programming (incremental erase of the chip takes quite a long time apparently). It would be possible to support both of these approaches though, if we delve deep into the DAPLink firmware. A flag in the HEX file could then allow us to switch between full-erase and partial-erase mode...

BLE FOTA flashing does support partial reflashes (this is what we currently already do, but only from SoftDevice upward. So this would still duplicate nrd sdk, mbed, microbit-dal etc. The Nordic bootloader causes a lot of issues to be honest - we may be better off with a new custom one that is less invasive if we're looking for something more advanced).

@mmoskal  This sounds like a good simple approach. The alternative would be something involving dynamic linking via a symbol table somewhere (but this would be a lot more complex). Leaving the hash up to the application software is a good idea i think, as it lets the app (e.g. PXT/Python) decide what the dependencies are.

This does bring up the question of what is inside the HEX files we create/ship around. Ideally, we'd want to pass partial HEX files (basically a diff) to the loaders. This would then also bring benefits of reduced download times - something very important for BLE channels in particular where the wireless link is the bottleneck. BUT, then the compilers would need to know the version of the runtime currently on the micro:bit, which sounds tricky, especially for the web based ones such as microbit.co.uk/PXT/mbed.org? 

Alternatively, we could always ship around full HEX files, and let the loaders decide which parts need reflashing. This probably wouldn't give much of a performance improvement though...

Joe

mmoskal, 2016-08-30:
I think the compiler doesn't need know - as long as we stick to, say, the structure I described at 0xff0, the flashing app on the phone could (somehow) read the existing one from the device and then decide what to send.

Maybe we could do the same for the USB - have the DAPLink look at the .hex file beforehand. For that, it would be useful for the hash to sit at the beginning of the file. I'm not sure if USB transfer or flashing is the bottleneck here, but USB should generally support more than 50k/s.

Now, that I think about it, it would probably make sense to have some magic number in front of the hash, so that we can distinguish between files that have it and ones that don't.

finneyj, 2016-08-30:
right - that could work. So compilers would always ship a full HEX file, and the loaders would decide on what needs reflashing based on the HASH values. 

In the case of BLE:
- this functionality is on the phone/tablet so avoids unnecessary data being sent over BLE.
- we could expose the hash over our device information service (or similar).

In the case of USB:
- we can read the flash directly of the nrf51822 from the interface chip to retrieve the hash
- perform a partial reflash if possible.

We would definitely need the HASH to be at the front of the HEX file somewhere. DAPlink can't cache a whole HEX file (not enough memory on the chip), so we'd need to parse it on the fly... 

The botttleneck for BLE programming is the BLE channel (by far). Bottleneck for USB is probably SWD - in which case a smaller HEX means faster reprogramming times.

mmoskal, 2016-08-30:
On the interface chip you could keep flashing as usual until you find the hash. If it matches you switch to a different mode. So you always re flash the few pages at the beginning but then you stop. This way there is no need to buffer anything. 
Still, even for this it's best to keep the hash at the beginning.

remay, 2016-08-30:
For the softdevice there is a version (FWID) and size in the softdevice information header at memory location 0x3000.   If you could store the first 3k of hex file you could compare the 16 bytes at that location, and if they are the same skip the softdevice flash, else re-flash from start of memory. (Talking about USB flash here - for BLE you'd need the BLE protocol to provide this data to the phone/tablet so it could do the same thing)

mmoskal, 2016-08-30:
Can you even flash the softdevice over bluetooth?

I guess most of the time in schools, the hex template won't change and we'll only have to flash 1-2k of code. In other cases it may be worth skipping softdevice, but the gain will be less (ie the flash will be slow anyway).

BTW 0x3000 is 12k.

remay, 2016-08-30:
> BTW 0x3000 is 12k

Indeed it is :-)

The flash memory map is (approximately):
MBR (0x00000000-0x00000FFF) 4k
SD (S110 0x00001000-0x00017FFF) 92k
App space (0x00018000-0x0003BFFF) 144k (inc. DAL, nordik and mbed layers)
Bootloader (0x0003C000-0x0003FFFF) 16k

At the moment (at least over USB) we clear down all 256k of flash, and re-write at least (4 + 92 +16) 112k plus our application.

Assuming the flash erase/write time is the limiting factor then avoiding re-writing the MBR/SD/Bootloader suggests that application flash time might be reduced by a factor of ~2 (and more for small applications.

I beleive that you can flash the softdevice (and the bootloader) over BLE already, but not the MBR:
https://devzone.nordicsemi.com/documentation/nrf51/6.0.0/s110/html/a00089.html#exp_ble_sdk_app_flash_operations_softdevice_sec
https://devzone.nordicsemi.com/documentation/nrf51/6.0.0/s110/html/a00089.html#exp_ble_sdk_app_flash_operations_application_sec

finneyj, 2016-08-30:
Thanks @remay - That looks the same as our analysis. 

A few observations about USB reflashing:

On nrf51822, we can either bulk erase the whole device, or erase block by block (block size is 1K). Each of those operations takes the same period of time (which I recall is in or around 20ms). so... doing a block by block erase all the time would be a bad idea, as it could increase programming time significantly (we have up to 256 blocks, hence an additional 5 or 6 seconds on a flash operation). Doing so selectively could save us a heap of time though.

I think we should look into a combined solution for USB - bulk erasing when a full reflash is needed, and a block by block rewrite when the relevant meta data is available. 

The main difficulty here is getting the meta data into the DAPLink software at the right time. We have no guarantee about the ordering of the file blocks that come in over USB, so putting the meta data at the front of the HEX file won't always work - blocks often arrive out of order. @mmoskal's idea of writing blocks as they arrive (even if not necessary!) might be the best we can do until the meta-data arrives...

remay, 2016-08-31:
USB flashing:   If I've extracted the right timings from the data sheet, and my assumptions hold up, then individual page erases and re-writng will be quick than bulk erase and re-write only if the number of pages to be written is less than 87 pages (i.e. an application image less than 87k).   At the moment a simple program with the DAL/nordic/mbed parts is ~65k - so this would be marginally faster;  about 400-500ms by my calculation.  My conclusion here is that it's not going to be worth the extra complexity unless we can reduce the number of pages that we need to flash significantly.

BLE flashing:   I suspect there are gains to be had here.   I assume that this must use page-by-page erase so that it doesn't erase the bootloader itself, which runs from flash.   So not only is this slower for the erase, but as the bootloader is running from flash the write times are considerably slower.    Getting the bootloader to run from RAM would help - but this may not even be possible due to it needing to run the softdevice for the bluetooth transfer.    I assume the OTA process is already only flashing the application image in single bank mode due to the sizes of our applications?  It's not something I've looked at.

mmoskal, 2016-09-01:
@remay I agree about application written in C++. When using a browser based compiler (in microbit.co.uk and PXT; possibly also with Python), the user code is in fact quite a bit smaller, as the runtime isn't consider part of it. For a typical few line user program it's going to be 1-2k. In that scenario, the .hex file looks like this:

* softdevice
* nordic/mbed runtime, DAL, C++ extensions/drivers
* user code (executable and compressed source)
* other data written in flash (file system etc)

mmoskal, 2016-09-05:
@finneyj I looked into the DAPLink code and I don't think they support blocks coming out of order: https://github.com/mbedmicro/DAPLink/blob/master/source/daplink/drag-n-drop/vfs_manager.c#L439

There are also some interesting oddities in intelhex.c - shadow_line is written to but never read, and when writing into the line buffer there is a size check but only in half of the cases https://github.com/mbedmicro/DAPLink/blob/master/source/daplink/drag-n-drop/intelhex.c#L196

mmoskal, 2016-09-05:
I think we could abuse the record type 2 or 3 of the .hex file (all the other ones are used). These would be normally ignored by the plain DAPLink software, but are still detected as .hex. We could put such a record in the first line of the .hex file, with the hash header and possibly an address where it's expected (0x1000-16 or so). Then, before calling `flash_manager_init()` (which normally erases the chip) we could read the target memory and compare the values. The `flash_manager_init()` isn't normally called until a chunk of the .hex file is read - the DAPLink tries to detect the type of the flash to be written - so there's already time to do that.

The actual hash value would be in both the 2/3 header at the top, and also in the .hex file, to make sure it's written correctly the first time.