lancaster-university/microbit-dal #208: First use of analog pin seems to reset period, resulting in wrong duty cycle

bluetooth-mdw, 2016-08-31:
The following test case, executed with pin 0 connected to an LED, should result in the LED flashing twice a second, then being switched off, then flashing again, then being switched off. The actual result though is that the first time the LED is switched on, it is solid (no flashing). The second time it flashes. 

```
   // first ON
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(511);
   
   uBit.sleep(3000);
   
   // OFF
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(0);

   uBit.sleep(3000);
     
   // second ON
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(511);
   
   uBit.sleep(3000);
   
   // OFF
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(0);
```

Switching the first two lines around resolves the problem however:

```
   // first ON
   uBit.io.pin[0].setAnalogValue(511);
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   
   uBit.sleep(3000);
   
   // OFF
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(0);

   uBit.sleep(3000);
     
   // second ON
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(511);
   
   uBit.sleep(3000);
   
   // OFF
   uBit.io.pin[0].setAnalogPeriodUs(500000);
   uBit.io.pin[0].setAnalogValue(0);
    

```

cpseager, 2016-08-31:
Maybe a null 'lastValue' the first time around here causes issues?

https://github.com/lancaster-university/microbit-dal/blob/c064eee3e7e34c2f0b745f37d87f96ea88779b98/source/drivers/DynamicPwm.cpp#L307

However I would say that value before period is the better order anyway.

bagofatoms, 2017-01-09:
I think its because the the pin has to be set as analog prior to setting its period, and that setAnalogvalue does just that.

pelikhan, 2017-01-09:
For servos, we recommend using pins.servoWritePin which uses setServoValue ( https://github.com/lancaster-university/microbit-dal/blob/5fa55812c7be1e428d9b255ca31b21cd66c72e4a/source/drivers/MicroBitPin.cpp#L248 )