lancaster-university/microbit-dal #210: Document why calibration is required to get the magnetic field strength

pelikhan, 2016-09-03:
Users ask... Do we really have to calibrate to get the magnetic field strength?

finneyj, 2016-09-03:
Really... yes, I'm afraid you do. It depends on the accuracy you want though. 

Most magnetometers (including the sensor on the micro:bit) are calibrated at time of manufacture, and contain an non-deterministic constant offset in each axis. They rely on the e-compass calibration algorithms to take out this error.  We *could* allow access to the raw data without calibration, but it wouldn't be accurate. 

Uncalibrated data would be ok for sensing relative values (i.e. changes in a field) within an axis though, but if you rotated the device around, the data would be just noise.

Thoughts?

finneyj, 2016-09-03:
suggestion: We already support multiple co-ordinate systems already (one of which is RAW), but RAW access to individual axis data may still triggers a calibration (although it doesn't need one):

https://github.com/lancaster-university/wellcome-dal/blob/master/source/drivers/MicroBitCompass.cpp#L429

We could enable that code path easily... which would provide a mechanism for users to take such measurements without being misleading as to the accuracy of the data?

pelikhan, 2016-09-04:
Private repo ☺

https://github.com/lancaster-university/wellcome-dal/blob/master/source/drivers/MicroBitCompass.cpp#L429

closed 2017-06-13 by alennie1993