lancaster-university/microbit-dal #212: MicroBitPin::getDigitalValue API cannot recognize the high level

ahnniu, 2016-09-22:
Hello,

We're working on some micro:bit accessories, We found there may be a bug in  [MicroBitPin::getDigitalValue](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitPin.cpp#L153) API. It cannot recognize the hign level. Microsoft Block Editor / Touch Devlop has this issue too.

### Issue

The [MicroBitPin::getDigitalValue](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitPin.cpp#L153) API read a digital value with an internal pull-down resistor configuration. But the internal pull-down resistance is only 13 kΩ, it’ll pull down the line voltage. In our case(our switch module has a 10K pull up resistor), it’s 1.9V, not 3.3V. So, the mcu cannot recognize the high level.

**Schematic Diagram**

[image]

### Sugguestions

Disable the internal pull-down resistor. Set the [MICROBIT_DEFAULT_PULLMODE](https://github.com/lancaster-university/microbit-dal/blob/master/inc/core/MicroBitConfig.h#L332) to PullNone.

### nRF51822 GPIO Specifications

[image]

### Source Code

https://github.com/lancaster-university/microbit-dal/blob/master/inc/core/MicroBitConfig.h#L332

```cpp
//
// Define the default mode in which the digital input pins are configured.
// valid options are PullDown, PullUp and PullNone.
//
#ifndef MICROBIT_DEFAULT_PULLMODE
#define MICROBIT_DEFAULT_PULLMODE                PullDown
#endif
```
https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitPin.cpp#L54

```cpp
/**
  * Constructor.
  * Create a MicroBitPin instance, generally used to represent a pin on the edge connector.
  *
  * @param id the unique EventModel id of this component.
  *
  * @param name the mbed PinName for this MicroBitPin instance.
  *
  * @param capability the capabilities this MicroBitPin instance should have.
  *                   (PIN_CAPABILITY_DIGITAL, PIN_CAPABILITY_ANALOG, PIN_CAPABILITY_AD, PIN_CAPABILITY_ALL)
  *
  * @code
  * MicroBitPin P0(MICROBIT_ID_IO_P0, MICROBIT_PIN_P0, PIN_CAPABILITY_ALL);
  * @endcode
  */
MicroBitPin::MicroBitPin(int id, PinName name, PinCapability capability)
{
    //set mandatory attributes
    this->id = id;
    this->name = name;
    this->capability = capability;
    this->pullMode = MICROBIT_DEFAULT_PULLMODE;

    // Power up in a disconnected, low power state.
    // If we're unused, this is how it will stay...
    this->status = 0x00;
    this->pin = NULL;

}
```

https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitPin.cpp#L153

```cpp
/**
  * Configures this IO pin as a digital input (if necessary) and tests its current value.
  *
  *
  * @return 1 if this input is high, 0 if input is LO, or MICROBIT_NOT_SUPPORTED
  *         if the given pin does not have digital capability.
  *
  * @code
  * MicroBitPin P0(MICROBIT_ID_IO_P0, MICROBIT_PIN_P0, PIN_CAPABILITY_BOTH);
  * P0.getDigitalValue(); // P0 is either 0 or 1;
  * @endcode
  */
int MicroBitPin::getDigitalValue()
{
    //check if this pin has a digital mode...
    if(!(PIN_CAPABILITY_DIGITAL & capability))
        return MICROBIT_NOT_SUPPORTED;

    // Move into a Digital input state if necessary.
    if (!(status & (IO_STATUS_DIGITAL_IN | IO_STATUS_EVENT_ON_EDGE | IO_STATUS_EVENT_PULSE_ON_EDGE)))
    {
        disconnect();
        pin = new DigitalIn(name, (PinMode)pullMode);
        status |= IO_STATUS_DIGITAL_IN;
    }

    if(status & (IO_STATUS_EVENT_ON_EDGE | IO_STATUS_EVENT_PULSE_ON_EDGE))
        return ((TimedInterruptIn *)pin)->read();

    return ((DigitalIn *)pin)->read();
}
```

### Attachments

- [nRF51822 Datasheet](http://infocenter.nordicsemi.com/pdf/nRF51822_PS_v3.1.pdf)

Best Regards,
Paul

finneyj, 2016-09-22:
hi @ahnniu 

Thanks for the detailed breakdown of this issue! Top marks from me. 

You're quite correct. This was noted here too in #156, and issues referenced from there. This atypical default originates from the BBC's testing/evaluation and a feature request for a default PullDown behaviour as they observed kids expecting inputs to read low when disconnected. Your observation that, as a result, hardware would need a stronger pull up, or disable the internal pull down is correct.

We try not to enforce policy decisions in microbit-dal, so the default can be specified in the build configuration options here, so that downstream languages/environments can defined the behaviour: 
https://github.com/lancaster-university/microbit-dal/blob/master/inc/core/MicroBitConfig.h#L332

The setPull() method will also let you define pull operations for an individual pin, and is exposed in python (I think)  and the PXT editors, but I don't believe made it into BlockEditor or TouchDevelop as BBC has the website under code freeze.

There will be an update to www.microbit.co.uk fairly soon, to bring on stream some accessibility support for the micro:bit. I'll ask the BBC is we can squeeze in a runtime update so that the setPull() method is present, and could be accessed via, say a TD library. We obviously must take care not to break assumptions made by other software and hardware about the current defaults.

@pelikhan Any further thoughts here re:  Block Editor / Touch Develop on microbit.co.uk?

pelikhan, 2016-09-22:
In order to update Blocks or Touch Develop, you would have to request the change to the BBC and they will prioritize it. As Joe said, PXT already supports this feature - in fact, PXT allows you to package and ship C++ as blocks, giving full control over the mbed stack.

ahnniu, 2016-09-23:
Thanks @finneyj and @pelikhan.  Really expect the new version of www.microbit.co.uk.

Hi @pelikhan, I saw you are the arthor of TouchDevelop and pxt-microbit. May I ask you a question here about TouchDevelop?

2 articles show users can create hybrid libraries in TouchDevelop:
- microbit-touchdevelop - [BITVM extensions](https://github.com/Microsoft/microbit-touchdevelop/blob/master/EXTENSIONS.md)
- An article from Jonathan Protzenko - [Use your BBC micro:bit to display time, temperature, and more!](https://jonathan.protzenko.fr/microbit/clock.html)

It shows there is a "use C++ compiler" checkbox can enable this function. But in my case there isn’t? What happens?  Does the microbit.co.uk accounts have different privilege level?  Thanks.

Best Regards,
Paul

[image]

finneyj, 2016-09-23:
@ahnniu no problem. I agree this needs fixing on microbit.co.uk.

This has been flagged to the BBC - hopefully they'll consider it soon.