lancaster-university/microbit-dal #213: Using serial on GPIO pins

kubaraczkowski, 2016-09-23:
Hi,

I seem not to be able to figure out how to operate the uart on GPIO pins. I've tried implementing a simple echo service (talking to a usb-uart dongle), but for the moment only pin 16 seems to respond (both as TX and RX). Tried pins 15,14,2, but none of them acts as it should. 
I tried first by having the "MicroBit uBit;" instance, then with simpler "MicroBitSerial serial()" component way - but no success. I do often see "clocks" on the non-working pins.
Any clue about that? Are only some GPIO pins able to act as uart pins? If so, which apart from pin 16?

Here's my very complex program :) 

```c++
#include "MicroBit.h"
MicroBitSerial serial(P0_15,P0_16);

int main(void) {

    serial.baud(9600);
   
    while(1) {     
        char x = serial.read();
        serial.sendChar(x);
        }
}
```

jamesadevine, 2016-09-24:
Hey!

You should not use P0_ as they are MCU pin mappings, we have a bunch of user mapped definitions here:

MICROBIT_PIN_

https://github.com/lancaster-university/microbit-dal/blob/master/inc/drivers/MicroBitPin.h

kubaraczkowski, 2016-09-24:
Oh crap, how silly of me not to find these mappings.

Perhaps that info could be made clear on the DAL documentation? The one
place I found it now is in the io section in example of setting a pin high
or low. Perhaps this could go somewhere close to the board/pin description
image? The names on the image (P0-P18) don't work and it's not obvious what
does.

Please compare with Arduino/esp8266/mbed pinout images where the pin name
on image matches a pin name in code.

finneyj, 2016-09-26:
It is odd that the mbed diagram and mbed target use the same naming convention, but with different meanings (one is the processor pinout, the other the edge connector)...

@kubaraczkowski  - I think improving documentation is the way to go here, but I worry a little about making things more confusing by having two diagrams... As you say, perhaps if we have a section explaining the difference, this would be fine.

kubaraczkowski, 2016-09-26:
Indeed, it's the similarity of naming conventions that's so misleading.
Perhaps defining the simplest "P0/P1/P2" (thus making mbed comply with the mbed diagram) could be a way out, though "p0/p1/p2" (lowercase) are already defined in PinNames.h ....

Additional documentation describing all the differences would be great - best to be placed at the level of uBit->Overview, and then referenced from io/uart/display pages.

The lack of complete schematic does not help here (or am I just unable to find it?), it's not easy to trace what is connecting to actual MCU pins. For example, tracing where LED rows go is not easy...

DavidWhaleIET, 2016-09-26:
Its mainly like the difference between BCM and BOARD numbering on the
Raspberry Pi. One set is the software peripheral in the chip, the other is
the hardware pin labeling on the board. I think Joe is right, its about
documenting it clearly. Signposts in appropriate header files also might
help.

Also I saw someone trip into this recently in the mbed IDE with the SPI.
The way we resolved this was by pointing at a small known working example.
So perhaps Joe we could check there are enough simple cookbook examples in
the API docs that show the correct method.

David.

finneyj, 2016-10-24:
OK - I've added another paragraph of detail into the uBit.io documentation. This should go live with the next release. 

I think we'll also revisit the mbed target soon too, and I'll see if we can inject some short Px names when we do.

closed 2016-10-24 by finneyj